ecdye/zram-config #8: Advantages of zram over "standard" /dev/ram0...

seamusdemora, 2019-04-27:
This isn't an issue per se, but a question. I've learned of your work here from [your post to a discussion](https://www.raspberrypi.org/forums/viewtopic.php?t=237735). My question is this: 

I understand the compression feature may be useful, but are there other advantages of **`zram-config`** (or **zram** in general) over the RAM disks currently allocated in Raspbian?

StuartIanNaylor, 2019-04-27:
As in tmpfs ?

Or a ramdisk?

tmpfs is sort of cool as its seems complete and will push idle pages to swap.
zram would if the writeback cache option wasn't disabled for some reason in raspbian the pi alarm arch linux for arm it works but even then it is only partition based and would seem it was supposed to offer more but stopped there.
Then again its idle control is manual so could be quite useful for mounted drives with a writeback cache as you mark all as idle and if accessed the idle mark is lost.
At stages you can push out to swap all still marked as idle.

Never tried it but that could be pretty cool in creating a zdrive where often files become old and static and just take space.

As a swap backed compressed ram zram is not as good as zswap but zswap is not enabled for the pi even though it is better.
Prob might make an appearance in buster with a bit of look.
To be honest not many actually understand zram & zram as the ubuntu package is an obvious example of that.
With zswap many say it should not be more than a 3rd of the memory my advise is to ignore what anyone says especially from ubuntu or mainline distro's ignore all and play with it yourself.
If it was anything to do with ubuntu we would all believe that it is still single stream and you need a zram device or zswap for each core which actually multi-stream was implemented in 3.15.

seamusdemora, 2019-04-27:
Yes - tmpfs vs ramdisk, good question! I was researching ramdisk (what is it used for), and the thread I linked to above showed up in my search result. At the time I read it, I was not clear on the difference between ramdisk and tmpfs. And so I assumed the thread was about ramdisk, and didn't read closely enough to realize my incorrect assumption. Bottom line: sorry for the confusion. 

But just so I'm clear: Is it true that `zram` and therefore `zram-config` are not applicable to ramdisk?

StuartIanNaylor, 2019-04-28:
Not sure at all why you are interested in ramfs as the newer tmpfs is likely a better alternative.
Zram is basically ramfs with compression but yeah ramdisk will not be included in zram-config as can not see the point but I have been thinking of including tmpfs as it has all the functions of ramfs but also the frontspace / frontswap system swap interface that still I can not remember its exact name.

seamusdemora, 2019-04-29:
I got interested in RAM disks (which is not exactly the same as ramfs) because I discovered that RAM disks are allocated in the Raspbian kernel, and no one seems to know why - or what these RAM disks are used for. 

I discovered they were there when I ran `fdisk -l`. I saw 16 RAM disks allocated: 

> Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
 ...

for ram0 - ram15. Since then, I've read that several Linux distros allocate RAM disks, but not all of the kernels *list* RAM disks in `/proc/partitions` (which is apparently why they show up in the `fdisk -l` output. Some have said listing them is erroneous kernel behavior. 

Allocating RAM disks would make sense (to me at least) if they were used to reduce write activity to the SD card. But that explanation is largely absent as their raison d'être - at least as far as I've read so far.  

And as far as ramfs vs tmpfs, I don't know enough yet to have an informed opinion regarding which is "best". That's actually why I'm researching this :) It's fascinating really - a bit like watching the inside of an anthill... all the ants working hard, but not really understanding the purpose of all of it. But I digress :)

StuartIanNaylor, 2019-05-01:
If you want a really unknowable account in /boot an image of the kernel+modules is created, initramfs and various other names depending on distro.
Arch linux is always a great reference https://wiki.archlinux.org/index.php/Mkinitcpio

closed 2019-05-11 by StuartIanNaylor