ecdye/zram-config #10: How do I use zram-config

spotty101, 2019-05-07:
How do I use zram-config   
I have changed the hard drive on an old laptop using mx linux ( just found out it does not use systemd ) to a ssd. 
So I thought I would try and reduce the writes to it  using zswap and found your project which also reduces log writes to the drive.

I have downloaded and installed using

sudo apt-get install git
git clone https://github.com/StuartIanNaylor/zram-config
cd zram-config
sudo sh install.sh

edited ztab to

# swap	alg	mem_limit	disk_size	swap_priority	page-cluster	swappiness
swap	lz4	200M		600M		75		0		90

# dir	alg	mem_limit	disk_size	target_dir		bind_dir
#dir	lz4	50M		150M		/home/pi		/pi.bind

# log	alg	mem_limit	disk_size	target_dir		bind_dir		oldlog_dir
log	lz4	50M		150M		/var/log		/log.bind		/opt/zram/oldlog

I started it using

sudo ./zram-config start
I got this error 
grep: /boot/cmdline.txt: No such file or directory

Starting zram

:~/zram-config
$ sudo ./zram-config start
[sudo] password for david: 
grep: /boot/cmdline.txt: No such file or directory

Laptop  No systemd

zramctl
NAME       ALGORITHM DISKSIZE DATA COMPR TOTAL STREAMS MOUNTPOINT
/dev/zram0 lz4           600M   4K   76B   12K       2 [SWAP]

df
Filesystem     1K-blocks      Used Available Use% Mounted on
udev             4001928         0   4001928   0% /dev
tmpfs             806188      1472    804716   1% /run
/dev/sdb2       19901264   6249376  12617904  34% /
tmpfs               5120         4      5116   1% /run/lock
tmpfs            2487680    183564   2304116   8% /run/shm
/dev/sdb3       80107640  19042820  56952440  26% /home
/dev/sda5       30865336  24375064   4915720  84% /mnt/MX-Data
/dev/sda4      323202812 162449304 160753508  51% /mnt/Windows
/dev/sda8       81258232  74505792   2601708  97% /mnt/Old-home
cgroup                12         0        12   0% /sys/fs/cgroup
tmpfs             806184        20    806164   1% /run/user/1000

free -h
              total        used        free      shared  buff/cache   available
Mem:           7.7G        3.5G        698M        563M        3.5G        3.4G
Swap:          4.6G          0B        4.6G

Note larger swap total

swapon
NAME       TYPE      SIZE USED PRIO
/dev/sdb4  partition 4.2G   0B   -2
/dev/zram0 partition 600M   0B   75

Note zram partition  I assume swap is working

but nothing suggesting log file has been moved

I am comparing with a virtualbox in a desktop with debian that uses systemd

here i can see the log seems to be set up

df shows
/dev/zram1 numbers /opt/zram/zram1
overlay1  number	/var/log

zramctrl
has extra line
/dev/zram1 lz4 number  /opt/zram/zram1

Thanks for your help

Dave

spotty101, 2019-05-07:
Thought I would try again without the shouting
sorry

edited ztab to
__# swap	alg	mem_limit	disk_size	swap_priority	page-cluster	swappiness
swap	lz4	200M		600M		75		0		90

__ #dir	alg	mem_limit	disk_size	target_dir		bind_dir
__#dir	lz4	50M		150M		/home/pi		/pi.bind

__# log	alg	mem_limit	disk_size	target_dir		bind_dir		oldlog_dir
log	lz4	50M		150M		/var/log		/log.bind		/opt/zram/oldlog

StuartIanNaylor, 2019-05-07:
zram-conf is very raspberry pi/arm specific.

https://github.com/StuartIanNaylor/zram-swap-config
https://github.com/StuartIanNaylor/zramdrive
https://github.com/StuartIanNaylor/log2zram
Are general individual services but still use systemd but guess we could just /etc/rc-local or /etc/inittab
The cmdline.txt alternative in non arm world would be a grub line I guess.

What distro and version are you using as I will knock up a branch for older x86

Are the more general ones but all use systemd as its so common now.
do a sh uninstall.sh but keep the repo as a source and we will mangle it.

Guess we can

spotty101, 2019-05-08:
I was wondering why you had the other 3. I thought you just decided to roll them into one.
Both the desktop and the laptop are debian stretch X64 based. 
The laptop is MXlinux which I just found out does not use systemD by default. I am assuming the destop does.

Would it be better if I uninstall and use the others?

Dave

StuartIanNaylor, 2019-05-08:
All systemd as much is now.
But will download Mxlinux

StuartIanNaylor, 2019-05-08:
MXLinux seems to be debian for those who don't like systemd as its actually there just not enabled.
The conversion of the script was simple really the emphemeral mode might take some more time but really never envisage for x86.
Glad its working and glad I also have a sysv-init version now.

closed 2019-05-11 by StuartIanNaylor