ecdye/zram-config #11: Please make ztab line For pi3

yutayu, 2019-06-09:
This is Great job. :)
But I can understand only disk_size.
I want to get 75% phisical memory.
How do I do ?

Thank you :)
Yutayuta

yutayu, 2019-06-09:
or default ztab is most efficnet for pi ?

StuartIanNaylor, 2019-06-09:
# swap	alg	mem_limit	disk_size	swap_priority	page-cluster	swappiness
swap	lz4	200M		600M		75		0		90

Disk size actually means very little as its the virtual size dependent on compression which is alg and input data = resultant compression and varies greatly.

mem_limit is the actual physical mem limit so if you have 1gb you could make that 750M.
Max virtual size is always a guess but with LZ4 x3 is about approx so 750*3 = 2250M

Usually 50% 500M mem_limit and 1500M disk size is enough but you can go with the above if you wish.

yutayu, 2019-06-10:
Thank you so much.
I don't close this issue , because I want to leave this perfect explanation.
I got it :) 
I try to change ztab disk_size 50% at first .
yutayuta

yutayu, 2019-06-10:
This script causes Good latency and fast :)

StuartIanNaylor, 2019-06-10:
The biggest thing is that it actually checks for other zram services at the start.
I still struggle to believe that simple method is missing from others and those scripts will not co-exist with other zram services.
That is even the packages in mainline distro's such as ubuntu & debian.
Then they still provide no control over mem_limit and disk size and have a alg to create a device per core and divide 50% ram between them.
zram has been multistream since 3.15 and the above is totally pointless and actually a worse solution especially with big/little core arrangement.
zram-config started as a bit of a protest project of read the docs!
https://www.kernel.org/doc/Documentation/blockdev/zram.txt

Then it got expanded a bit to use zram disks to cover high write dirs and protect against sd card block wear.
You might want to create a zlog as that is also quite beneficial on a Pi.

yutayu, 2019-06-11:
Much thanks :)

closed 2019-06-11 by yutayu

yutayu, 2019-07-01:
I also use this.
https://github.com/graysky2/profile-sync-daemon
According to this , It uses tmpfs 

[image]

as overlaid ram.
Now your script uses fixed size ram.
I think tmpfs is better.
How  ?

StuartIanNaylor, 2019-07-11:
tmps should when idle push out to swap so if zram is installed then essentially it will be doing the same thing.
They only advantage is that zram-config is 100% compressed if idle or live.
I will have a look one time