ecdye/zram-config #12: Permission error occurs.

yutayu, 2019-06-10:
# dir	alg	mem_limit	disk_size	target_dir		bind_dir
dir	lz4	50M		150M		/home/pi		/pi.bind

pi@raspberrypi:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       184G   13G  164G   7% /
devtmpfs        433M     0  433M   0% /dev
tmpfs           438M  219M  220M  50% /dev/shm
tmpfs           438M   12M  426M   3% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           438M     0  438M   0% /sys/fs/cgroup
tmpfs           438M   48K  438M   1% /tmp
/dev/sda1        43M   23M   21M  53% /boot
/dev/zram1      130M   24M   96M  20% /opt/zram/zram1
overlay1        130M   24M   96M  20% /home/pi
tmpfs            88M     0   88M   0% /run/user/1000

When I use hexchat, It occurs.

StuartIanNaylor, 2019-06-10:
Yeah I got that before when I did the whole /home/pi directory but though I had sorted the permissions error.
Quick fix is to be more specific with the Dir as zlog is the same but you will get no error there.
I will have a look but will have to root out a Pi and image before I get round to starting.

yutayu, 2019-06-11:
I got it :)
But Great job for pi.

yutayu, 2019-06-11:
Overlayfs sometime causes disks error, it needs fsck and we can fix it.
I used profile-sync-daemon,sometimes it occurs, and zram-config , too.
Do you know any idea to avoid disk error?

yutayu, 2019-06-11:
PARTUUID=xxxxx  /               ext4    defaults,noatime,lazytime,commit=60  0       1

This is my fstab.

StuartIanNaylor, 2019-06-11:
Yeah I had to sleep on it and I don't think we are going to get anywhere until 4.19 as I think loads of overlayfs updates that will fix it went in 4.18.
haven't got round to checking yet been side tracked with something else, that I am struggling to finnish.

yutayu, 2019-06-11:
Okay.
This problem may be because of my environment.
Thanks :)

closed 2019-06-11 by yutayu