ecdye/zram-config #13: zram dir read-only

mstormi, 2019-06-18:
I have one zram directory that cannot be written to (read only errors)
dir     lz4     50M             300M            /var/lib/openhab2       /openhab2.bind
The others work fine meanwhile.
log     zstd    50M             250M            /var/log                /log.bind               /volatile/oldlogs
dir     lz4     10M             1000M           /var/cache              /cache.bind

Any idea ? I *think* it was writable for some time in the beginning after boot. Eventually the difference is that the amount of data written is larger than the mem_limit - could that be an explanation ?

StuartIanNaylor, 2019-06-19:
mem_limit is uncompressed actually memory use. As long as the it managed to compress into the mem_limit approx x3 then it should fit.

I think its overlayfs and I have stopped being active for a couple of months and much is forgotten.
I got the feeling in 4.17 or 4.18 a load of overlayfs patches went into the kernel and thought 4.19 was imminent and never bothered, but 4.19 and testing has taken much longer than I expected.
I think its that or I am doing something stupid with the initial bind.

mstormi, 2019-06-21:
mem_limit of the incriminated directory x3 = 150M was still less than that directory took on disk.
I encountered this on a second occasion (the /var/cache shown above) whose mem_limit was obviously way to low, and it started showing after apt-get update (the probably only program to result in writes to that dir).
I've now increased mem_limit of the first dir and removed the /var/cache line.
No problems since. Will keep you updated here.
FWIW, I'm using kernel 4.14.98.

Will you keep supporting/keep this code up to date ? I've just created [this PR](https://github.com/openhab/openhabian/pull/576) to use your wonderful creation in openHABian.
Please keep up the great work.

Another thing, could you please explain how the syncing and move to oldlogs mount works ?
Most important thing I'm asking myself is how often the sync will take place and when rotated logs will be moved to oldlogs. A pointer to existing docs will suffice.

StuartIanNaylor, 2019-06-24:
Its a compromise of gaining the CoW copy up on write of OverlaysFS which always confuses me as it does have kernel inclusion but in comparison to Aufs has a near lack of tools.
It uses https://github.com/kmxz/overlayfs-tools merge down but requires the overlayfs to be remounted as its likely the merge layer upper & lower will become out of sync.
You gain huge reductions in memory usage as only writes are pulled in zram but syncing needs to be offline.
I have asked the overlayfs guys a hey "you supplying any tools guys?" but the answer seems to be no.
I have thought about maybe doing a Aufs version.

closed 2019-07-21 by StuartIanNaylor