ecdye/zram-config #14: syncToDisk ?

mstormi, 2019-06-23:
I noticed the syncToDisk command is commented out  in zram-config
What does it mean, will changes to upper layer FS (in ram) ever get synced to lower layer (SD card)?
Can I enforce a sync ?

StuartIanNaylor, 2019-06-24:
Uses mergeoverlay synctodisk doesn't exist copying whole directories on shutdown is just a bad idea.

mstormi, 2019-06-24:
Ok, but as mergeOverlay() is only called on zram-config stop, files will only be merged when that is executed ? I'd like to sync once a day or so, how about adding a "sync" option to zram-config ?

This would execute the following, right ? it doesn't work although I _am_ root
[08:48:08] root@devpi:/home/pi# /usr/local/lib/zram-config/overlay merge -v -u /var/log -l /opt/zram/log.bind
The program cannot write trusted.* xattr. Try run again as root.
[08:48:12] root@devpi:/home/pi#

StuartIanNaylor, 2019-06-24:
You can not as you have to merge offline so cronjob reboot or work out a way to umount / from a working system.

mstormi, 2019-07-09:
Just an idea, I have come across https://github.com/amir73il/overlayfs/wiki/Overlayfs-snapshots
Would that work to snapshot first and then (still online) copy the snapshot ?

StuartIanNaylor, 2019-07-11:
This is the weird thing as supposedly no as the man himself Amir seemed to say.
Have a look in the issue marked 'Help'

Also would you try if my check for kernel 4.19 or less splits correct now and install Attr or Xattr depending on the previous

mstormi, 2019-07-11:
Yeah read that. What I understood is that 
- snapshots are available in overlayfs (eventually need to be activated in the kernel)
- Amir the overlayfs guy has never used the [overlay tools](https://github.com/kmxz/overlayfs-tools)
My interpretation is: noone tried to combine snapshots and those tools so it's worth a try ?

StuartIanNaylor, 2019-07-12:
Please do mstormi see how it works. I have been playing with Wayland/Panfrost of late so not able to do any prolonged testing.

This is what I don't get as the most perfect example of snapshots would be able to do a oldest snapshot merge down and cycle through a history to any point.
I am totally bemused by the lack of overlayfs tools in this regard and that makes me think there must be a reason for this that I am not understanding as I am like you.

The 'glue' in the middle layer is being used to merge down the lower with my limited understanding but nothing removes that glue and therefore surely it becomes out of sync?
Maybe it does merge down and clean the merge layer its been a while since I really examined with depth.

Then there is another problem of race conditions of any file access in the upper of source files in the lower and middle merge.
If you could lock the upper file as it is processed then prob nirvana but just have a hunch overlayfs tools is far simpler and just goes through the whole layer. Creates serialised cp, mv & rm and permission functions and just runs that as a script at the end.

Its a huge race scenario that is hugely likely to miss current activity and I am still unsure about merge layer update after completion.

I would be absolutely amazing to have live merges on an overlayfs and if you would whilst I am faffing with wayland give it a go?

closed 2019-07-21 by StuartIanNaylor