ecdye/zram-config #15: Doesn't work with Raspbian Buster Lite

Felixrising, 2019-06-24:
sudo sh install.sh 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  libattr1-dev
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 33.8 kB of archives.
After this operation, 79.9 kB of additional disk space will be used.
Get:1 http://mirror.datamossa.io/raspbian buster/main armhf libattr1-dev armhf 1:2.4.48-4 [33.8 kB]
Fetched 33.8 kB in 1s (27.5 kB/s)       
Selecting previously unselected package libattr1-dev:armhf.
(Reading database ... 38598 files and directories currently installed.)
Preparing to unpack .../libattr1-dev_1%3a2.4.48-4_armhf.deb ...
Unpacking libattr1-dev:armhf (1:2.4.48-4) ...
Setting up libattr1-dev:armhf (1:2.4.48-4) ...
Processing triggers for man-db (2.8.5-2) ...
Cloning into 'overlayfs-tools'...
remote: Enumerating objects: 92, done.
remote: Total 92 (delta 0), reused 0 (delta 0), pack-reused 92
Unpacking objects: 100% (92/92), done.
gcc -Wall -std=c99 -c main.c
In file included from main.c:16:
/usr/include/linux/stat.h:56:8: error: redefinition of ‘struct statx_timestamp’
 struct statx_timestamp {
        ^~~~~~~~~~~~~~~
In file included from /usr/include/arm-linux-gnueabihf/sys/stat.h:446,
                 from main.c:14:
/usr/include/arm-linux-gnueabihf/bits/statx.h:25:8: note: originally defined here
 struct statx_timestamp
        ^~~~~~~~~~~~~~~
In file included from main.c:16:
/usr/include/linux/stat.h:99:8: error: redefinition of ‘struct statx’
 struct statx {
        ^~~~~
In file included from /usr/include/arm-linux-gnueabihf/sys/stat.h:446,
                 from main.c:14:
/usr/include/arm-linux-gnueabihf/bits/statx.h:36:8: note: originally defined here
 struct statx
        ^~~~~
make: *** [makefile:11: main.o] Error 1
install: cannot stat 'overlayfs-tools/overlay': No such file or directory
Created symlink /etc/systemd/system/basic.target.wants/zram-config.service → /etc/systemd/system/zram-config.service.
#####          Reboot to activate zram-config         #####
#####       edit /etc/ztab to configure options       #####

StuartIanNaylor, 2019-06-28:
yeah xattr has changed to attr
in install.sh change to git clone -b fix_xattr_lib_include https://github.com/Izual750/overlayfs-tools and that should work

StuartIanNaylor, 2019-06-28:
I guess I will just have to detect kernel version and distro and swap between the 2.

yutayu, 2019-07-01:
I am using zram-config and /dir and zram swap at raspbian buster desktop.
It works.

Felixrising, 2019-07-01:
modifying install.sh to git clone -b fix_xattr_lib_include https://github.com/Izual750/overlayfs-tools works. I tried the overlayfs-tools build with and without, with -b fix_xattr_lib_include was the trick.

StuartIanNaylor, 2019-07-04:
Cheers thought it would the lib name changes with 4.19 I have a pi4 and just waiting for a PSU so will edit the script so it will auto select.

Felixrising, 2019-07-06:
I had this error on both a RPI3 and RPI4, seems to be common to Buster:-

cat /proc/device-tree/model 
Raspberry Pi 3 Model B Rev 1.2
lsb_release  -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

cat /proc/device-tree/model
Raspberry Pi 4 Model B Rev 1.1
lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

yutayu, 2019-07-06:
Or How about trying to use lzo instead of lz4?
Firefox-esr crashed often with lz4. 
I am testing lzo with firefox and now fine.

yutayu, 2019-07-06:
firefox-esr crashed again lol

but , network is smoother than lz4.
lzo is good for 3B + raspbian.

StuartIanNaylor, 2019-07-07:
For those using zram dir for home will have to test. Got the pi4 but been busy but will have a look at buster.
It must be something to do with permissions as if you set ephemeral mode and make all r/o then all is ok.
So if you can do that you should be able to do it on a single dir even if it is ~/

StuartIanNaylor, 2019-07-11:
Would you check that guys?

Just did a kernel check and if 4.19 then Attr if less then Xattr

closed 2019-07-16 by StuartIanNaylor