ecdye/zram-config #18: xattr version not building

mstormi, 2019-07-22:
pi@devpi:/tmp/z $ git clone -b fix_xattr_lib_include https://github.com/Izual750/overlayfs-tools
Klone nach 'overlayfs-tools' ...
remote: Enumerating objects: 96, done.
remote: Total 96 (delta 0), reused 0 (delta 0), pack-reused 96
Entpacke Objekte: 100% (96/96), Fertig.
pi@devpi:/tmp/z $ cd overlayfs-tools/
pi@devpi:/tmp/z/overlayfs-tools $ make
gcc -Wall -std=c99 -c main.c
gcc -Wall -std=c99 -c logic.c
logic.c: In function ‘is_opaquedir’:
logic.c:38:32: error: ‘ENOATTR’ undeclared (first use in this function)
     if ((res < 0) && (errno != ENOATTR)) {
                                ^~~~~~~
logic.c:38:32: note: each undeclared identifier is reported only once for each function it appears in
makefile:14: die Regel für Ziel „logic.o“ scheiterte
make: *** [logic.o] Fehler 1

StuartIanNaylor, 2019-07-24:
```
version=$(uname -r)
major=`echo $version | cut -d. -f1`
minor=`echo $version | cut -d. -f2`

if [ "$major" -ge "4" ]
then
        if  [ "$minor" -ge "19" ]
	then
		InstallAttr
	else
		if [ "$major" -ge "5" ]
        	then
                	InstallAttr
        	else
                	InstallXattr
        	fi
	fi
else
		InstallXattr
fi
```
Should install the correct version depending on stretch or buster? if its >= 4.19

mstormi, 2019-07-24:
Yes stretch is on 4.19.57. Just dist-upgraded to make sure it's up to date.
Note it has no /usr/include/<sys>.
Meanwhile xattr.h to define ENOATTR is still in <attr>. Dunno why lzual750 changed that to sys ?
Either way wouldn't it make sense to NOT include his repo (just clone the standard one) and add #ifndef ENOATTR ... in *your* code ?
If I'm not mistaken that's the only difference, or am I overlooking something there ?

[09:52:59] root@devpi:/usr/include# grep ENOATTR attr/*
attr/xattr.h:#ifndef ENOATTR
attr/xattr.h:# define ENOATTR ENODATA        /* No such attribute */
[09:52:59] root@devpi:/usr/include#

Btw. does that [ "$major" -ge "5" ] make sense ?

StuartIanNaylor, 2019-07-24:
Yes as I am running 5.2 and it also works on other distro's.
https://en.wikipedia.org/wiki/Raspbian as its not only me who finds strange that a LTS based scheduled distro should jump LTS and not just dist-upgrade.
Anyway I will just check the lib exists.
Try that just edited the git haven't tried but should work.

mstormi, 2019-07-24:
Hmm, you still clone and try compiling the modified tools repo. That still fails.

BTW, please (when done) create a tag v0.2 or whatever so I can clone that instead of master.

pi@devpi:~/openhabian/z $ sudo sh install.sh
        libattr.so.1 (libc6,hard-float) => /lib/arm-linux-gnueabihf/libattr.so.1
        libattr.so (libc6,hard-float) => /lib/arm-linux-gnueabihf/libattr.so
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
libattr1-dev ist schon die neueste Version (1:2.4.47-2).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
Klone nach 'overlayfs-tools' ...
remote: Enumerating objects: 96, done.
remote: Total 96 (delta 0), reused 0 (delta 0), pack-reused 96
Entpacke Objekte: 100% (96/96), Fertig.
gcc -Wall -std=c99 -c main.c
gcc -Wall -std=c99 -c logic.c
logic.c: In function ‘is_opaquedir’:
logic.c:38:32: error: ‘ENOATTR’ undeclared (first use in this function)
     if ((res < 0) && (errno != ENOATTR)) {
                                ^~~~~~~
logic.c:38:32: note: each undeclared identifier is reported only once for each function it appears i                    n
makefile:14: die Regel für Ziel „logic.o“ scheiterte
make: *** [logic.o] Fehler 1
install: der Aufruf von stat für 'overlayfs-tools/overlay' ist nicht möglich: Datei oder Verzeichnis                     nicht gefunden
Synchronizing state of zram-config.service with SysV service script with /lib/systemd/systemd-sysv-i                    nstall.
Executing: /lib/systemd/systemd-sysv-install enable zram-config
#####          Reboot to activate zram-config         #####
#####       edit /etc/ztab to configure options       #####
pi@devpi:~/openhabian/z $

StuartIanNaylor, 2019-07-24:
PS for the repo so you have your static version that you can adopt to fit as you wish

closed 2019-07-24 by StuartIanNaylor

mstormi, 2019-07-24:
I mirrored your repo and will take my clones from there

Your current install.sh first line is broken. But once corrected, it built and runs.

yutayu, 2019-07-25:
root@raspberrypi:/home/pi/Documents/zram-config# sh install.sh
install.sh: 4: install.sh: Syntax error: "(" unexpected (expecting "}")

StuartIanNaylor, 2019-07-27:
@mstormi I also decided to clone both the original overlayfs tools so that then I can edit if need be.
There is still difference between stretch & buster and both are needed using /etc/os-release is a much better solution than kernel but presumed stretch would stay 4.14

Thanks