ecdye/zram-config #19: master version is not building.

yutayu, 2019-07-25:
root@raspberrypi:/home/pi/Documents/zram-config# sh install.sh
install.sh: 4: install.sh: Syntax error: "(" unexpected (expecting "}")

I cannot install zram-config in raspbian buster.

pi@raspberrypi:~/Documents/zram-config $ uname -r
4.19.58-v7+

StuartIanNaylor, 2019-07-25:
Yeah my bad I thought I was working in a branch and updated master earlier its the first line from mem
} to the end

yutayu, 2019-07-25:
still it occurs same erros.
I want to try to use it -> IRQ line

StuartIanNaylor, 2019-07-25:
I have not got a Pi setup check tomoz as it should be a simple fix but will need to run.

yutayu, 2019-07-25:
before this change, we can install this to pi.
https://github.com/StuartIanNaylor/zram-config/commit/90f2bbfc84583673cf2c2b141d51fc7af2032c14#diff-3fbb47e318cd8802bd325e7da9aaabe8

StuartIanNaylor, 2019-07-25:
The line 1 error should be clear just rm the zram-config dir clone again and try

yutayu, 2019-07-25:
pi@raspberrypi:~/Documents/zram-config $ sudo sh install.sh
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libattr1-dev is already the newest version (1:2.4.48-4).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Cloning into 'overlayfs-tools'...
remote: Enumerating objects: 14, done.
remote: Counting objects: 100% (14/14), done.
remote: Compressing objects: 100% (14/14), done.
remote: Total 110 (delta 5), reused 0 (delta 0), pack-reused 96
Receiving objects: 100% (110/110), 40.44 KiB | 154.00 KiB/s, done.
Resolving deltas: 100% (61/61), done.
gcc -Wall -std=c99 -c main.c
In file included from main.c:16:
/usr/include/linux/stat.h:56:8: error: redefinition of ‘struct statx_timestamp’
 struct statx_timestamp {
        ^~~~~~~~~~~~~~~
In file included from /usr/include/arm-linux-gnueabihf/sys/stat.h:446,
                 from main.c:14:
/usr/include/arm-linux-gnueabihf/bits/statx.h:25:8: note: originally defined here
 struct statx_timestamp
        ^~~~~~~~~~~~~~~
In file included from main.c:16:
/usr/include/linux/stat.h:99:8: error: redefinition of ‘struct statx’
 struct statx {
        ^~~~~
In file included from /usr/include/arm-linux-gnueabihf/sys/stat.h:446,
                 from main.c:14:
/usr/include/arm-linux-gnueabihf/bits/statx.h:36:8: note: originally defined here
 struct statx
        ^~~~~
make: *** [makefile:11: main.o] Error 1
install: cannot stat 'overlayfs-tools/overlay': No such file or directory
#####          Reboot to activate zram-config         #####
#####       edit /etc/ztab to configure options       #####

I don't hurry :)
Good Night.

StuartIanNaylor, 2019-07-25:
Yeah looks like my push to the wrong branch I will have to just run up a Pi in the morn.
Thanks though and apols

yutayu, 2019-07-26:
No error occurs :)
Perfect.

closed 2019-07-26 by yutayu

StuartIanNaylor, 2019-07-27:
Apols about that I was working in the wrong branch but actually made me use /etc/os-release which is far better.
I just wish the same libs with the same functionality would stop swapping function removals I guess its all optimisation and tidying but actually my c knowledge is extremely minimal. :)

Shame about IRQs on the PI until I tried I would of never expected affinity can not be set.