ecdye/zram-config #20: Bug - dir and log versions are not preserving ownership and permissions of target_dir

sislakd, 2019-08-04:
The easiest way to demonstrate the issue is to configure zram overlay for /tmp folder:

`dir     lz4     150M            400M            /tmp                    /tmp.bind`

Once zram-config performs configuration, newly created /tmp directory for mount is created with default ownership and permissions which are root:root 0755. This cause issues for any process running under different than root permission. This process cannot create/write into /tmp at all. The only possible workaround is to have sub-directory with broader permissions already created in /tmp and reconfigure temp path for all non-root processes.

The expected functionality should be: within createZdir mkdir for TARGET_DIR should preserve ownership and permission from original which is at that moment remounted to ${ZDIR}${BIND_DIR}.

StuartIanNaylor, 2019-08-04:
Don't configure zram for tmp. Haven't a clue how zram would bind to a tmpfs folder but never occured to me that it would be of any use at all.

StuartIanNaylor, 2019-08-04:
tmpfs when idle pushes out to zswap so this is a pointless exercise

sislakd, 2019-08-04:
/tmp is not always tmpfs, there are many systems where /tmp is just directory in the root partition like in raspbian. Next, /tmp was just example, the described bug occurs on any directory as the problem is with not preserved ownership and permissions. Simply, the current implementation works only for root or if there is a higher level directory from which it takes proper default ownership. Which is limiting factor.

StuartIanNaylor, 2019-08-04:
Yeah you are right as really perms should be cloned I will get round to that sometime this week.

closed 2019-08-05 by StuartIanNaylor