ecdye/zram-config #21: New feature - support for in ram volumes only

sislakd, 2019-08-04:
It would be nice to have ability to support in ram volumes only within /etc/ztab. For example, there can be introduced new mount type ram (or tmp) which will have one parameter less compared to dir (bind_dir is not needed). This will result into zram volume creation and direct mount without overlayFS into target_dir. Such option would be beneficial for the cases where some mount doesn't need to be backed by disk at all, e.g. /tmp.

It is possible to configure to create own startup scripts to achieve desired behavior, however it would be very nice to have it configurable at one place /etc/ztab.

StuartIanNaylor, 2019-08-04:
If its not going to be persistent then just create a zram drive via zramctl.

closed 2019-08-04 by StuartIanNaylor

sislakd, 2019-08-04:
I thought that this zram-config project was exactly about zram configuration and overlayfs was just one function. Thus, this request was about addition of support for zram without overlayfs.

StuartIanNaylor, 2019-08-04:
There is no need for a pointless regression that has no benefit.
You can always clone copy fork and provide yourself but in all honesty my opinion is that its a pointless option.