ecdye/zram-config #22: Improvement - oldlog_dir flatterns logging structure and mix logs together

sislakd, 2019-08-04:
For log configuration within /etc/ztab, zram-config creates automatically simple logrotate configuration defining global logdir option pointing to oldlog_dir. This works perfectly for logs inside target_dir and while some log is rotated it is automatically offloaded to non-overlay directory. However, many applications (e.g. apache, nginx) creates sub directories within /var/log folder and configures logrotate configuration for those sub directories. Currently, the simple 00_oldlog implies that rotated logs from sub directories are stored directly into oldlog_dir. This introduces confusion as many times log filenames are the same for different services (e.g. error.log). Logs named using the same filename are mixed together into oldlog_dir within one file sequence (e.g. error.log.1 is from apache, error.log.2 is from haproxy, error.log.3 is again from apache and error.log.4 is from nginx).

It would be very helpful, if createZlog can: 
- enumerate directory structure within original content of target_dir
- configure appropriate logrotate olddir option for every directory

Thus, logs from apache2 sub directory will be rotated into ${oldlog_dir}/apache2 and so on. This will maintain desired log structure as in the original log location.

StuartIanNaylor, 2019-08-04:
That isn't zlog its logrotate.
https://github.com/logrotate/logrotate maybe make suggestion there.

closed 2019-08-04 by StuartIanNaylor

sislakd, 2019-08-04:
Ok, then logrotate configuration which is done by zram-config is useless and an user have to configure it properly by himself and use just dir option. log option in /etc/ztab is not usable due to described issues with its too simple configuration.

StuartIanNaylor, 2019-08-04:
Its logrotate if you wish a more complex configuration then configure logrotate.
The config of how you wish to provide for complex log organisations has nothing to do with the base zram-config functionality.
It has absolutely nothing to do with zram-config, zram-config provides a simple way to provide a logrotate.conf that ships off that block device to save ram.
		echo "olddir "${OLDLOG_DIR} > /etc/logrotate.d/00_oldlog
		echo "createolddir 755 root root"  >> /etc/logrotate.d/00_oldlog
		echo "renamecopy" >> /etc/logrotate.d/00_oldlog

Thats it.

If you want a more complex one leave 'oldlog_dir' blank and create the conf yourself

sislakd, 2019-08-04:
I don't want complex behavior. I thought, that your zram-config was supposed to be used for /var/log where it is normal to have log files structured within multliple sub-directories. I haven't changed anything, just installed standard packages like apache and zram-config makes it confusing. 

I somewhere read that you would like to have suggestions to improve zram-config to make it usable. But no problem, will create own solution which will help in standard installations. Now, zram-config is useful just for compressed swap. dir is unusable due to wrong ownership and permissions (rejected #20 ). log additional functionality (which is oldlog_dir) must be configured separately anyway. And if you want save SD card/SSD writes it is recommended to have swap disabled (even with small Raspberry PI, there is enough RAM to what you need, no need for swap for standard usage). Thus, not sure what is benefit of zram-config usage.

StuartIanNaylor, 2019-08-05:
lols you got me on a particularly bad day, something is screwy with my efi on my old I7 and ended up wiping all and restarting.
zramconfig just uses logrotate to move old logs out of zram and can run with or without.
The focus is to stop block wear on sd/flash/nand not log organisation.
The functionality is there and its logrotate and highly complex log manipulations can be achieved.

zram-config started off just as a protest that te distro versions are extremely poor.
First thing they do is assume zram is just swap, worse by omitting simple checks it will not co-exist with other zram services.
Then being beneficial zlog & zdir got added to basically make the statement that zram is not just swap.
Even now most distro's don't add the compile option for the writeback cache which completely confuses me as even if the module option is enabled you still have to enable its use.

Its not zram-config that ships out logs as said its log-rotate and zram-config just makes it easy to enable a basic logrotate.conf to do that.
Its logrotate you need to make advanced log configurations with and it is out of scope with zram-config part from the is function to move out logs from zram by another application.

logrotation is clearly logotates remit and the name is an indication.
Maybe it was the permissions that should be now fixed even if a bit hacky just to chown/chmod after mount