ecdye/zram-config #26: Write-error on swap-device

Seketh, 2019-09-27:
Running Raspian Buster on a RPi4 4GB.

Zram-config works brilliantly until one or two days of uptime, when dmesg gets flooded with "write-error on swap-devices". Here's a sample of the last few lines:

`
[101734.099250] Write-error on swap-device (254:0:1134984)
[101734.099300] Write-error on swap-device (254:0:1135016)
[101734.099420] Write-error on swap-device (254:0:1135088)
[101734.099473] Write-error on swap-device (254:0:1135104)
[101734.099526] Write-error on swap-device (254:0:1135120)
[101734.099577] Write-error on swap-device (254:0:1135136)
[101734.099631] Write-error on swap-device (254:0:1135152)
[101734.101754] Write-error on swap-device (254:0:1135448)
[101734.339315] systemd-journald[119]: /dev/kmsg buffer overrun, some messages l                                                                                    ost.
[101734.794086] oom_reaper: reaped process 581 (rclone), now anon-rss:0kB, file-                                                                                    rss:0kB, shmem-rss:0kB`

I'm using the following (although default settings show the same error): 

`NAME       ALGORITHM DISKSIZE   DATA  COMPR  TOTAL STREAMS MOUNTPOINT
/dev/zram1 lz4           450M 427.3M    92M  96.2M       4 /opt/zram/zram1
/dev/zram0 lz4           1.5G 271.7M 124.3M 128.5M       4 [SWAP]
`
I have deactivated the swap and am only using zram.

yutayu, 2019-09-27:
Same error 3B on raspbian without adding diskswap.
I add diskswap , and no error happens.

pi@raspberrypi:~ $ swapon
NAME       TYPE      SIZE   USED PRIO
/var/swap  file      1.8G     0B   -2
/dev/zram0 partition 1.3G 519.8M   75

Seketh, 2019-10-13:
Wasn't that obvious at first, this issue happens when mem_limit is full. 

How to avoid this? What's the point of disk_size?

yutayu, 2019-10-13:
Maybe, This is just only lack of swap.

StuartIanNaylor, 2020-08-12:
Is this still happening?

yutayu, 2020-08-12:
Aug 12 05:20:19 raspberrypi kernel: Read-error on swap-device (254:0:573072)
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.4.51-v7+ #1327 SMP Thu Jul 23 10:58:46 BST 2020 armv7l GNU/Linux

StuartIanNaylor, 2020-08-13:
Post you ztab as likely your disk_size is to high a ratio to mem_limit or you have idle pages that are already compressed maybe or its a bug with lz4/zram.

To be honest the Ztab defaults are extremely optimistic as the original focus was logs and text logs are highly compressible.
Try 2x disk_size:mem_limit and if works maybe up to 2.5x disk_size:mem_limit or you have found a bug and should via Raspberry.

yutayu, 2020-08-13:
# Configuration file for zram-swap-config (https://github.com/StuartIanNaylor/zram-swap-config) under GNU General Public License v3.0 license.
# This configuration file is read by the zram-swap-config service

# MEM_FACTOR = Percentage of available ram to allocate to all zram swap devices which is divided equally by swap_devices number
MEM_FACTOR=75
# DRIVE_FACTOR = Virtual uncompressed zram drive size = estimate approx alg compression ratio an results and is the % of the Memfactor
DRIVE_FACTOR=210
# COMP_ALG lz4 is faster than lzo but some distro's show compile and operation difference and in use lzo depending on binary may be faster
COMP_ALG=lzo-rle
# SWAP_DEVICES = number of indivial drives sharing memeory provided by MEM_FACTOR each device support multiple streams 1 large drive is generally better
SWAP_DEVICES=1
# SWAP_PRI = swap_priority for each drive 75 is a hig order preference and should be well above other swap drives
SWAP_PRI=75
# PAGE_CLUSTER default page cluster is 3 which caches fetches in batches of 8 and helps with HDD paging, with zram mem 0 forces single page fetches
# This can help reduce latency and increase performance
PAGE_CLUSTER=0
# SWAPPINESS default swappiness is 60 but with increased performance of zram swap 80 garners overall performance gain without excessive load
# Because zram uses compression load is created and even if minimal at intense load periods such as boot any extra load is unwanted
# Unfortunately there is no dynamic load balancing of swappiness as with zram in general operation SWAPINESS=100 will garner performance benefit
# If the overall load is reasonable at high load it will cause load to accumulate. 
SWAPPINESS=90
# If you check my repo there will also be a simple dynamic load branch based SWAPPINESS governor that will get of the hurdle of a static compromise on swappiness

sorry, mine is zram-swap-config.

yutayu, 2020-08-13:
I changed DRIVE_FACTOR=210 DRIVE_FACTOR=250.

yutayu, 2020-08-13:
NAME       ALGORITHM DISKSIZE  DATA COMPR TOTAL STREAMS MOUNTPOINT
/dev/zram0 lzo-rle       1.6G  828K   74B    4K       4 [SWAP]

This works well,sometimes error occurs like above.

StuartIanNaylor, 2020-08-14:
I was going to say your swapping repos's.

Yeah its your drive factor probably as even if 210% seems like its struggling to compress.
Unless its lzo-rle as was going to say now we are on 5.4 its in LZO-RLE is in the kernel.

The is always deflate but boy that is slow (zlib) zstd is a trade off that does give higher compression with not so bad compress and decompress speed.

I have been trying for a while to get Raspberry to enable the write back cache which is a safe guard against this where very infrequently we will dump to SD if idle in zram for a long time.
Yeah I know it is writes but very low infrequent writes that you have control over, but for now that doesn't solve things.

Because resultant compression is just pure guesswork and that the compression level of input is totally dependent on what your software is doing, you are going to have to drop the drive_factor to as low as 125% as lower it becomes a bit pointless.
Do it in 20% steps and see how you go.

It is interesting you are seeing decompress errors though and maybe this is something introduced in 5.4 and lzo-rle as I have only had problems compressing the uncompressible, aka already compressed.

yutayu, 2020-08-14:
root@raspberrypi:/home/pi# zramctl
NAME       ALGORITHM DISKSIZE DATA COMPR TOTAL STREAMS MOUNTPOINT
/dev/zram0 zstd        820.5M   4K   58B    4K       4 [SWAP]

Maybe I was too greedy.

yutayu, 2020-08-14:
MEM_FACTOR=50
DRIVE_FACTOR=125

root@raspberrypi:/home/pi# zramctl
NAME       ALGORITHM DISKSIZE DATA COMPR TOTAL STREAMS MOUNTPOINT
/dev/zram0 zstd          547M   4K   58B    4K       4 [SWAP]

StuartIanNaylor, 2020-08-15:
OK but that is only a 125% compression rate so build up.
The amount of ram you allocate prob should run in 75%
`free -h` should give you a good idea.

Its not greedy its just the unknown that a random app might hold a large chunk of compressed data in memory and be idle.
Its not that usual but it can happen and you just have to tune a bit by getting to a working the work up.

Is zstd working ok as never did try it as its not blazing fast like lzo-rle but guess maybe a good compromise

yutayu, 2020-08-15:
root@raspberrypi:/home/pi# free -h
              total        used        free      shared  buff/cache   available
Mem:          875Mi       324Mi        77Mi        16Mi       473Mi       475Mi
Swap:         546Mi        18Mi       528Mi

As router, It seems fine.
I will try it and to increase mem_factor and ratio as desktop.

yutayu, 2020-08-15:
root@raspberrypi:/home/pi# zramctl
NAME       ALGORITHM DISKSIZE  DATA  COMPR TOTAL STREAMS MOUNTPOINT
/dev/zram0 zstd          1.5G  2.1M 622.9K  1.1M       4 [SWAP]

root@raspberrypi:/home/pi# free -h
              total        used        free      shared  buff/cache   available
Mem:          875Mi       345Mi        84Mi        22Mi       445Mi       449Mi
Swap:         1.4Gi       2.0Mi       1.4Gi

MEM_FACTOR=75
DRIVE_FACTOR=225

zstd compression ratio is 2.8 , so I set DRIVE_FACTOR around 2.8.

yutayu, 2020-08-15:
I misunderstood.

compression ratio is <3. so DRIVE_FACTOR should be <150% .

StuartIanNaylor, 2020-08-15:
Google on there Chromebooks ran 50% memory 200% compression (drive factor)  which should be achievable.
Only thing now is load as presume on boot things rattle a bit but should calm down.
Slower compression is usually because its more intensive (load) but guess you can lower the SWAPPINESS but the we we are employing zram is because we wish to use it :)

See how you go but yeah drive_factor is likely the important one but also check mem usage with `free -h`.
Usually no tuning is needed but seems you might be running something quite load and memory hungry.

yutayu, 2020-08-17:
Raspbian Desktop Use.
It seems no problem.
But dmesg left some lines.

root@raspberrypi:/home/pi# free -h
              total        used        free      shared  buff/cache   available
Mem:          874Mi       542Mi        75Mi        86Mi       255Mi       189Mi
Swap:         1.3Gi       463Mi       847Mi
 
[58257.470735] Backtrace: 
[58257.470756] [<8010d480>] (dump_backtrace) from [<8010d778>] (show_stack+0x20/0x24)
[58257.470763]  r6:b33d6000 r5:00000000 r4:80d95cb8 r3:baec4fd4
[58257.470774] [<8010d758>] (show_stack) from [<808cdbe4>] (dump_stack+0xe0/0x124)
[58257.470788] [<808cdb04>] (dump_stack) from [<802c7324>] (warn_alloc+0xd4/0x15c)
[58257.470795]  r8:80d05684 r7:00000a00 r6:80ab60e4 r5:00000000 r4:80d04f48 r3:baec4fd4
[58257.470803] [<802c7250>] (warn_alloc) from [<802c84f4>] (__alloc_pages_nodemask+0x1090/0x123c)
[58257.470807]  r3:00000000 r2:80ab60e4
[58257.470813]  r6:00000000 r5:00000000 r4:00000200
[58257.470821] [<802c7464>] (__alloc_pages_nodemask) from [<802c8940>] (page_frag_alloc+0x180/0x1a0)
[58257.470828]  r10:00000000 r9:00000000 r8:00000000 r7:00000a00 r6:00000a20 r5:00000b40
[58257.470831]  r4:b37a0e48
[58257.470844] [<802c87c0>] (page_frag_alloc) from [<8077b35c>] (__netdev_alloc_skb+0x150/0x184)
[58257.470849]  r7:00000a00 r6:b37a0e48 r5:b2b6e000 r4:00000b40
[58257.470857] [<8077b20c>] (__netdev_alloc_skb) from [<80694084>] (rx_submit+0x10c/0x23c)
[58257.470863]  r7:00000a00 r6:a916fcc0 r5:ad83cb80 r4:b2b6e540
[58257.470868] [<80693f78>] (rx_submit) from [<80694ba4>] (rx_complete+0x200/0x220)
[58257.470875]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:a916fcc0 r5:ad83cb80
[58257.470879]  r4:b2b6e540 r3:00000001
[58257.470887] [<806949a4>] (rx_complete) from [<8069ecec>] (__usb_hcd_giveback_urb+0x74/0x118)
[58257.470892]  r7:00000000 r6:00000000 r5:b335ee00 r4:ad83cb80
[58257.470900] [<8069ec78>] (__usb_hcd_giveback_urb) from [<8069eddc>] (usb_hcd_giveback_urb+0x4c/0xfc)
[58257.470906]  r7:80d04f48 r6:ad83cb80 r5:ad83cb80 r4:b0e26ec0
[58257.470916] [<8069ed90>] (usb_hcd_giveback_urb) from [<806ca854>] (completion_tasklet_func+0x90/0xd0)
[58257.470922]  r6:ad83cb80 r5:b324b800 r4:b0e26ec0 r3:60000113
[58257.470930] [<806ca7c4>] (completion_tasklet_func) from [<806d9fec>] (tasklet_callback+0x20/0x24)
[58257.470935]  r7:b37962c4 r6:00000000 r5:b29fd7c4 r4:b29fd7c0
[58257.470946] [<806d9fcc>] (tasklet_callback) from [<8012628c>] (tasklet_action_common+0x64/0xec)
[58257.470955] [<80126228>] (tasklet_action_common) from [<8012633c>] (tasklet_hi_action+0x28/0x30)
[58257.470961]  r9:00000001 r8:b305e000 r7:00000100 r6:b33d6000 r5:00000000 r4:80d03080
[58257.470972] [<80126314>] (tasklet_hi_action) from [<8010249c>] (__do_softirq+0x1a4/0x418)
[58257.470979] [<801022f8>] (__do_softirq) from [<80126130>] (irq_exit+0x100/0x110)
[58257.470986]  r10:b925173c r9:b33d6000 r8:b305e000 r7:00000001 r6:00000000 r5:00000000
[58257.470989]  r4:80ca42e4
[58257.471000] [<80126030>] (irq_exit) from [<80181200>] (__handle_domain_irq+0x70/0xc4)
[58257.471007] [<80181190>] (__handle_domain_irq) from [<80102228>] (bcm2836_arm_irqchip_handle_irq+0x60/0xa4)
[58257.471016]  r8:ac4d800d r7:b33d76f4 r6:ffffffff r5:20000013 r4:00000000 r3:b33d76c0
[58257.471022] [<801021c8>] (bcm2836_arm_irqchip_handle_irq) from [<80101a3c>] (__irq_svc+0x5c/0x7c)
[58257.471026] Exception stack(0xb33d76c0 to 0xb33d7708)
[58257.471033] 76c0: ac4d800d 00001ff3 00000000 0000000b 80d04f48 00000553 b918ff38 b33d7840
[58257.471040] 76e0: ac4d800d b92a9b3c b925173c b33d7774 00001ff3 b33d7710 000000ff 7f20d298
[58257.471043] 7700: 20000013 ffffffff
[58257.471047]  r4:7f20d298 r3:baec4fd4
[58257.471084] [<7f20d1ec>] (HUF_compress4X_repeat [zstd_compress]) from [<7f229a58>] (ZSTD_compressBlock_internal+0x260/0x12d0 [zstd_compress])
[58257.471091]  r10:7f22f2a0 r9:b92a9b3c r8:00000000 r7:00000004 r6:00000553 r5:80d04f48
[58257.471093]  r4:b918f008
[58257.471119] [<7f2297f8>] (ZSTD_compressBlock_internal [zstd_compress]) from [<7f22acb0>] (ZSTD_compressContinue_internal+0x1e8/0x43c [zstd_compress])
[58257.471126]  r10:ac4d8006 r9:00001ffa r8:953e1000 r7:00001000 r6:00001000 r5:b918f008
[58257.471129]  r4:953e2000
[58257.471154] [<7f22aac8>] (ZSTD_compressContinue_internal [zstd_compress]) from [<7f22b938>] (ZSTD_compressCCtx+0x2f8/0x628 [zstd_compress])
[58257.471161]  r10:32c30001 r9:00000004 r8:00000001 r7:00000008 r6:b33d7a00 r5:b918f074
[58257.471164]  r4:b918f008
[58257.471184] [<7f22b640>] (ZSTD_compressCCtx [zstd_compress]) from [<7f1b312c>] (__zstd_compress+0x88/0xa0 [zstd])
[58257.471190]  r10:00002000 r9:b918f008 r8:00001000 r7:953e1000 r6:ac4d8000 r5:b33d7b28
[58257.471193]  r4:00000000
[58257.471203] [<7f1b30a4>] (__zstd_compress [zstd]) from [<7f1b31b4>] (zstd_compress+0x34/0x3c [zstd])
[58257.471210]  r10:b33d6000 r9:00000000 r8:80e24fb8 r7:80d07d44 r6:b33d6000 r5:7f1d5400
[58257.471213]  r4:b0e3aa00
[58257.471226] [<7f1b3180>] (zstd_compress [zstd]) from [<8051f970>] (crypto_compress+0x2c/0x34)
[58257.471244] [<8051f944>] (crypto_compress) from [<7f1d1218>] (zcomp_compress+0x40/0x48 [zram])
[58257.471264] [<7f1d11d8>] (zcomp_compress [zram]) from [<7f1d2a3c>] (zram_bvec_rw+0x3e8/0xa50 [zram])
[... 29 lines cut ...]
[58257.471464] 7fa0:                                     00000000 00000000 00000000 00000000
[58257.471470] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[58257.471476] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[... cut ...]
[58257.471585] Normal: 1050*4kB (UMEHC) 691*8kB (UEH) 320*16kB (UEH) 85*32kB (UE) 15*64kB (UE) 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 18528kB
[58257.471667] 12324 total pagecache pages
[58257.471678] 22 pages in swap cache
[58257.471688] Swap cache stats: add 2050607, delete 2050690, find 8606/1407398
[58257.471694] Free swap  = 447308kB
[58257.471701] Total swap = 1342940kB
[58257.471708] 229376 pages RAM
[58257.471715] 0 pages HighMem/MovableOnly
[58257.471722] 5552 pages reserved
[58257.471727] 16384 pages cma reserved

yutayu, 2020-08-17:
journalctl -xe

Aug 17 23:09:01 raspberrypi kernel: [<8051f944>] (crypto_compress) from [<7f1d1218>] (zcomp_compress+0x40/0x48 [zram])
Aug 17 23:09:01 raspberrypi kernel: [<7f1d11d8>] (zcomp_compress [zram]) from [<7f1d2a3c>] (zram_bvec_rw+0x3e8/0xa50 [zram])
Aug 17 23:09:01 raspberrypi kernel: [<7f1d2654>] (zram_bvec_rw [zram]) from [<7f1d3168>] (zram_rw_page+0xc4/0x15c [zram])
Aug 17 23:09:01 raspberrypi kernel: [<7f1d30a4>] (zram_rw_page [zram]) from [<8033caf0>] (bdev_write_page+0x8c/0xc8)

yutayu, 2020-08-17:
I chenged swappiness=90 to 100.
and trying to use.

StuartIanNaylor, 2020-08-18:
swapiness is just pref of use, so higher means swap will be used even if more mem is avail.

As to above dunno? But looking like zstd is favoured either.

yutayu, 2020-08-19:
I didn't know.
I got it.
I got swappiness=80.

yutayu, 2020-08-20:
I set it back to 100; 100 is less error-prone than 80.

yutayu, 2020-08-20:
I found this article.
https://haydenjames.io/linux-performance-almost-always-add-swap-part2-zram/

we shuld set sysctl for zram.

StuartIanNaylor, 2020-08-20:
Its already done in zram-swap-config.conf
Have you looked at the code?
```
	sysctl vm.page-cluster=${PAGE_CLUSTER}
	sysctl vm.swappiness=${SWAPPINESS}
```

Also just so others know we are talking about zram-swap-config.

How 100 can be less error prone than 80 is very confusing as your forcing more use the higher the number.
But the problem seems to be zram that you have, all zram-swap-config does is config the zram.
You will have to ask the zram maintainers about zram errors.

closed 2020-08-20 by StuartIanNaylor

yutayu, 2020-08-21:
I alerady changed sysctl.conf.
I restored it.
and I trust you :)

yutayu, 2020-08-27:
MEM_FACTOR=50
seems good.

yutayu, 2020-09-12:
Failure to match algorithm compression and disk size results in an error.
For example, if you select zstd, double the disk size is not enough, so you need to make it nearly triple.

yutayu, 2020-09-21:
With raspbian desktop, It seems only lz4 is certain choice.
Because of something error with others.
Maybe it depends on its speed.

StuartIanNaylor, 2020-09-23:
There is no disk and double the disk size is basically guessing low compression, its mem_factor that allocates ram usage and smaller disks are just a lower expectation on overall compression.
There are certain operations that just fail as already compressed huge chunks that can not be compressed fill past the available ram.
Its rare but it does happen seems from experience only when compiling certain apps.
If something is not working with zram and has changed file a bug on the raspbian desktop repo version.

yutayu, 2020-10-31:
This is my memo.
System looks at disk_size , not original mem_limit.

yutayu, 2020-11-18:
@Seketh 
https://github.com/raspberrypi/linux/issues/3839

StuartIanNaylor, 2020-11-22:
So you have a 8gb Pi 4 ?

yutayu, 2020-11-22:
Me ? no . 
This is pi kernel problem.

StuartIanNaylor, 2020-11-25:
That problem is only with the 8gb Pi4

yutayu, 2020-11-25:
No.Several error with zram is reported.

yutayu, 2020-12-05:
https://github.com/armbian/build/pull/2377

StuartIanNaylor, 2020-12-05:
That is armbian and a kernel option any absolutely nothing to do with this project. Choose or compile your own kernel but those choices are not part of this repo.

yutayu, 2020-12-05:
@StuartIanNaylor 
Excuse me , I thought that I showed you related info.
But here is github.

StuartIanNaylor, 2020-12-05:
Post on the kernels or https://www.kernel.org/ is likely to have better effect. This is just an app that obs need a kernel and uses zram but isn't the source...

yutayu, 2020-12-06:
@StuartIanNaylor 
I agree with you.
and I leave it a while.

StuartIanNaylor, 2020-12-06:
I don't know what is happening but seems its a kernel option that needs to change like you posted. Guess its just time where new kernels will adopt and disable the malloc method that is the source.

yutayu, 2020-12-11:
@StuartIanNaylor 
I see.
and Have you never seen these decompression error  around your pi?

StuartIanNaylor, 2020-12-11:
Yes and the open source code that creates it is not in this repo but others and where you should be posting rather than spamming.

I have handed over the repo, so no longer have to be polite.

yutayu, 2020-12-11:
@StuartIanNaylor 
Another repo you mean is raspberrypi kernel repo?

StuartIanNaylor, 2020-12-11:
its ecdye/zram-config so you can post as you wish.

yutayu, 2020-12-11:
@StuartIanNaylor 
Okay.
I think that you did great job some japanese rely on your script.
:)

yutayu, 2020-12-11:
and I should make post and no spam anymore :P