ecdye/zram-config #32: Who wants the project transferred to?

StuartIanNaylor, 2019-10-30:
I am not developing this any more haven't picked up a Pi or SBC for months.
So if someone wants to take over please do.

The main problem is lack of tools with OverlayFS and the strange nature that OverlayFS is a default module in most distro's.
Aufs is the opposite has great tools but would require kernel mods...

I really thought by now we would have merge tools especially with their work with snapshots as to be able to rotate snapshots down to lower would be an awesome backup tool.

But hey and apols for being absent but just not involved anymore.

ecdye, 2020-09-07:
@StuartIanNaylor I am willing, there are several improvements I have made that can be seen in #42

StuartIanNaylor, 2020-09-10:
Actually no, apols but think that might be a bad idea.

Just take the code and make your own, but to add another service just to sync a service seems extremely over the top.
It would seem to me that your adding extras because there is still file access on a file service and what you should be doing is stopping that file access not adding extra here.

Zram-config is already far too bloated and complex for what it is, its no problem as the initial protest with poor utils from the big distro's was to show zram was more than just a badly implemented swap file.
We should be asking the OverlayFS team for better merge down tools maybe and making this all far simpler.
You guys at openhab do whatever you wish for openhab and how you want to implement but I think I will pass on this offer.

closed 2020-09-10 by StuartIanNaylor