ecdye/zram-config #34: How about adding a fomula to allocate memory like this?

yutayu, 2020-01-05:
https://github.com/novaspirit/rpi_zram/blob/master/zram.sh

StuartIanNaylor, 2020-01-07:
There is no need to split via cores as zram is multistream and all cores can access a single zram device.
You can add as much ram to zram as you wish and because application varies so much there isn't a formula apart from 50-25% is prob a good initial guestimate.
Disk size is very dependent on alg chosen to expected compression.

yutayu, 2020-01-08:
I am sorry, I mean this line.
mem=$(( ($totalmem / $cores)* 1024 ))
for example, 50% disksize
mem=$(( ($totalmem /2)* 1024 ))

now,we can set disksize as we like without fomula.

StuartIanNaylor, 2020-01-08:
You can but zram-config runs a ztab so multiple devices and disks can be setup.
Its zram-config which has multiple uses for zram and not just zram-swap.
If it was just zram-swap then I would agree

yutayu, 2020-01-08:
https://github.com/StuartIanNaylor/zram-swap-config
MEM_FACTOR=50

I mean this .very sorry lol

closed 2020-01-08 by yutayu