ecdye/zram-config #36: How to make a drive that will not save back to the original location?

kingmomomo, 2020-03-15:
How to make a drive that will not save back to the original location? I would like to make a copy of the original location but not wish to save it back. Thanks.

rogerdahl, 2020-03-16:
In `/usr/local/bin/log2zram`, comment out the call to `mergeOverlay` so that it looks like:
```
# mergeOverlay >>${ZLOG} 2>&1 || exit 1
```

StuartIanNaylor, 2020-03-22:
It uses a mergeOverlay tool https://github.com/kmxz/overlayfs-tools which as far as I know is the only tool for overlayfs.

Apart from https://github.com/https://github.com/amir73il/overlayfs/wiki/Snapshots-HOWTO/overlayfs/wiki/Snapshots-HOWTO

So as roger said systemctl edit zram-config and stop the ExecStop=/usr/local/bin/zram-config stop from being run by editing it.

I guess create a snapshot that you can mount elsewhere but apart from knowing of its existence that is the limit of my snapshot knowledge.
I was always hoping amir73il was going to do a merge down from lastest snapshot so you could rotate down through a history.
I would have a read and see if it will provide what you need.

Fork the project or just copy and maybe create a merge elsewhere version?

closed 2020-08-12 by StuartIanNaylor