ecdye/zram-config #37: Should we consider using zram-config for new projects

prakashkrishna, 2020-03-18:
Hi StuartIanNaylor,

According to your post below, you're not actively developing zram-config.
https://github.com/StuartIanNaylor/zram-config/issues/32#issue-514928350
Please let us know if zram-config can be considered for new projects. If not, do you have any suggestion for an alternative?

Thanks,
Prakash

StuartIanNaylor, 2020-03-22:
Its sort of complete Prakash and should be fine.
I am just not active as its a small script just filling in some of the holes left by similar scripts supplied by main distro's.
There is no dev-work really as merge tools should be really supllied by overlay-fs and the rest is just zramctl.

It is what it is a demonstration that the main distro's should upgrade thier zram scripts so zram services can be interoperable and optimised.
If someone want to take ownership they can but as far as I am aware the main distro scripts are still hobnailed with the first attempt script of zrams first release.

There are a few others but also as far as I am aware they all miss elements of interoperability or optimisation included here.

closed 2020-03-22 by StuartIanNaylor

prakashkrishna, 2020-03-23:
Thanks for clarifications StuartIanNaylor!