ecdye/zram-config #38: Confused about page-cache setting

wanton7, 2020-05-03:
I'm no Linux expert and I tried to search information about page-cache setting that you mention in README.me and at https://github.com/StuartIanNaylor/zram-config/tree/master/swap-performance .
But I couldn't find any setting named like that by googling. How do I set page-cache to 0 in Ubuntu?

StuartIanNaylor, 2020-05-04:
Already does it for you in the ztab
```
# swap	alg	mem_limit	disk_size	swap_priority	page-cluster	swappiness
swap	lz4	250M		750M		75		0		80
```
page_cluster is set to 0 by default.

wanton7, 2020-05-04:
Ok thanks, I didn't understand page-cache settings means vm.pagecluster. From reading documents about it, I thought it only effects reads from swap not page writes.

closed 2020-05-04 by wanton7