ecdye/zram-config #41: doesn't sync on reboot

mstormi, 2020-08-09:
I have problem of zram data *sometimes*tr to disappear on reboot. 

I've debugged no instance where the lazy unmount apparent didn't work and the routine returned without calling mergeOverlay()
Now as the shutdown sequenced proceeded, data was lost.

Known problem? Is it ok to remove the ||return 1 behind the umount .... -l ?

StuartIanNaylor, 2020-08-12:
Its a race condition that MergeOverlay should always be run unmounted, so MergeOverlay says.
I have waiting patiently for OverLayFS official tools, but nothing happening.

Its going to be a catch 22 if not unmounted then data could be lost as it will be written after.
It will not be merged down because we could not unmount prob due file access.

I didn't write MergeOverlay maybe you can ask which is worst / best scenario

closed 2020-08-12 by StuartIanNaylor

mstormi, 2020-08-24:
While the mergeOverlay problem may still exist or not (there were recent changes on the overlayfs-tools), it wasn't the problem here.

The problem was that the upper FS actually can get unmounted on shutdown BEFORE zram-config.service gets executed.
I've setup a script to store / restore zram dir contents on start(boot) / stop (reboot) of the zram-sync service but it didn't sync
upper FS contents to a permanent storage on disk. Reason being that the zram dirs get (or actually *can* get) unmounted before 
store ... on reboot/final.target, systemd sends a stop to ALL running services and it's coincidence whether that happens before or after zram-config/stop . I didn't find out which service is the culprit but I found a way around.
The key is to insert RequireMountsFor=/opt/zram/zram1/upper or similar into zram-config.service 
Will provide a patch once I can confirm it's working.