ecdye/zram-config #43: Question: Does zram-config work together with fail2ban?

smokemyshoes, 2020-09-22:
Hi

Maybe a totall noob question but nevertheless.... Will fail2ban work after using log2ram?

If yes, will it require any (re-)configuration of fail2ban?

Many thanks in advance!

StuartIanNaylor, 2020-09-23:
I guess its just the service order and that fail2ban should start after zram-config as that means it will close before and not create log locks on shutdown.

But answer is without trying there should be no reason why not as the logs still retain the same path but are on a bind mount.
If you have any problems maybe edit the service and put zram-config in the after section of the of the fail2ban service.

smokemyshoes, 2020-09-23:
Thanks. I'll give it a try.

closed 2020-09-23 by smokemyshoes