ecdye/zram-config #50: Bash Script to Update zram-config

BroMarduk, 2020-12-16:
A script to update zram-config to the latest version would be nice.   This prevents having to do an uninstall followed by an install that overwrites the ztab configuration.

Would something like this work?  (It seems to for me, but I'm not sure if I missed anything important...)

```
#!/usr/bin/env bash

if [[ "$(id -u)" -ne 0 ]]; then
  echo "ERROR: You need to be ROOT (sudo can be used)."
  exit 1
fi
if [[ ! -f /usr/local/sbin/zram-config ]]; then
  echo "ERROR: zram-config is not installed."
  exit 1
fi

if ! dpkg -s 'make' 'libattr1-dev' &> /dev/null; then
  apt-get install --yes make libattr1-dev || exit 1
fi

if [[ $(systemctl is-active zram-config.service) == "active" ]]; then
  echo "ERROR: zram-config service is still running. Please run \"sudo systemctl stop zram-config.service\" to stop it and then uninstall before running this."
  exit 1
fi

cd overlayfs-tools || exit 1
make
cd ..

echo "Disabling zram-config service."
systemctl disable --now zram-config.service

install -m 755 zram-config /usr/local/sbin/
install -m 644 zram-config.service /etc/systemd/system/zram-config.service

if [[ ! -f /etc/ztab ]]; then
  install -m 644 ztab /etc/ztab
fi

if [[ ! -d /usr/local/share/zram-config/log ]]; then
  mkdir -p /usr/local/share/zram-config/log
fi

install -m 644 uninstall.bash /usr/local/share/zram-config/uninstall.bash
install -m 644 ro-root.sh /usr/local/share/zram-config/ro-root.sh

if [[ ! -f /etc/logrotate.d/zram-config ]]; then
  install -m 644 zram-config.logrotate /etc/logrotate.d/zram-config
fi

install -m 755 overlayfs-tools/overlay /usr/local/lib/zram-config/overlay

systemctl enable --now zram-config.service

echo "#####          zram-config has been updated.          #####"
echo "#####       edit /etc/ztab to configure options       #####"
```

ecdye, 2020-12-16:
That is pretty close, a few things need tweaking. I'll work one adding a service to update it, thanks for the suggestion

BroMarduk, 2020-12-16:
Awesome.  It's very possible I messed something up, since the RPi3 takes 5 minutes to shut down.   Thanks for looking.

commit f1f79839 (2020-12-16): Add update script

Fixes: #50
commit 52fea0b7 (2020-12-16): Add update script

Fixes: #50
closed 2020-12-16 by ecdye