ecdye/zram-config #51: Five Minute Shutdown of Service

BroMarduk, 2020-12-17:
Using a new Buster 5.4.79 (December 2020) image.   
Default Lite image, with latest package updates.

Installed git package
Installed zram-config

No changes to ztab config file

If rebooting/shutting down , the shutdown hangs at "A stop job is running for zram-config (xxs / 5min)"

If stopping using sudo zram-config "stop" the hang occurs at: ` systemctl stop systemd-journald.socket systemd-journald-audit.socket systemd-journald-dev-log.socket` for about 5 minutes.

Is this normal behavior?  Am I missing some package in the Lite version?

BroMarduk, 2020-12-17:
[image]

ecdye, 2020-12-17:
Sort of, this is caused by the issues mentioned in the README relating to the systemd ordering dependencies. If it becomes an issue, you can manually stop zram-config before rebooting.

commit d716edc0 (2020-12-28): Fix shutdown service

Fixes: #51
closed 2020-12-28 by ecdye