ecdye/zram-config #56: bad persistence on ZRAM

mstormi, 2020-12-30:
just FYI, I ran into an extraordinarily mean error on my own production system that took me half a day to find out
(with shutters all down and heating off ☹️ ):

Some .rrd file (rrd4j persistent storage) was broken and that made Java crash repeatedly.
But deleting it didn't help. Persistence files are on ZRAM but seems when I deleted it ZRAM took the older version from the lowerFS and seems *that* was the broken one.
Duh.

Not I don't know yet what to make of that, just documenting for now.

Maybe one thing is we should extend the permissions_corrections() apply to /opt/zram/*.bind/* too (the lowerFS copies of
all the ZRAM'ed dirs).

ecdye, 2020-12-30:
This may have been caused by the improper sync of zram that was happening on shutdown until Sunday. I don't think it is a major issue.

mstormi, 2020-12-30:
until Sunday ?
Are you aware that zram-config, unlike openhabian, does not get updated on every openhabian-config start ?
There (still) multiple reports of reboot to take very long or "fail".

ecdye, 2020-12-31:
Yes I am aware. They need to update. See my comment of the other thread.

mstormi, 2020-12-31:
ok let's close it given we assume with your fix and my permissions PR there's nothing left to do. We can reopen if I'm wrong here.

closed 2020-12-31 by mstormi