ecdye/zram-config #58: This script is broken at raspbian.

yutayu, 2021-02-07:
I install this script at raspbian and run .
Try it please.

uname -a
Linux raspberrypi 5.10.11-v7+ #1399 SMP Thu Jan 28 12:06:05 GMT 2021 armv7l GNU/Linux

yutayu, 2021-02-07:
zram-config doesn't mount depending on ztab correctly.

yutayu, 2021-02-07:
pi@raspberrypi:~ $ journalctl -xe | grep zram
Feb 08 02:26:35 raspberrypi systemd[1]: Started zram-config.
-- Subject: A start job for unit zram-config.service has finished successfully
-- A start job for unit zram-config.service has finished successfully.
Feb 08 02:26:37 raspberrypi zram-config[520]: + TMPDIR=/tmp
Feb 08 02:26:37 raspberrypi zram-config[520]: + ZSHARE=/usr/local/share/zram-config
Feb 08 02:26:37 raspberrypi zram-config[520]: + ZLOG=/usr/local/share/zram-config/log/zram-config.log
Feb 08 02:26:37 raspberrypi zram-config[520]: + ZDIR=/opt/zram
Feb 08 02:26:37 raspberrypi zram-config[520]: + [[ '' == \s\h\u\t\d\o\w\n ]]
Feb 08 02:26:37 raspberrypi zram-config[520]: + case "$1" in
Feb 08 02:26:37 raspberrypi zram-config[520]: ++ date +%Y-%m-%d-%H:%M:%S
Feb 08 02:26:37 raspberrypi zram-config[520]: + echo 'zram-config start 2021-02-08-02:26:37'
Feb 08 02:26:37 raspberrypi zram-config[520]: + rm -fv /tmp/zram-device-list.new
Feb 08 02:26:37 raspberrypi zram-config[520]: + file=/etc/ztab
Feb 08 02:26:37 raspberrypi zram-config[520]: + ZTAB_EMPTY=true
Feb 08 02:26:37 raspberrypi zram-config[520]: + read -r line
Feb 08 02:26:37 raspberrypi zram-config[520]: + case "$line" in
Feb 08 02:26:37 raspberrypi zram-config[520]: + continue
Feb 08 02:26:37 raspberrypi zram-config[520]: + read -r line
Feb 08 02:26:37 raspberrypi zram-config[520]: + case "$line" in
Feb 08 02:26:37 raspberrypi zram-config[520]: + continue
Feb 08 02:26:37 raspberrypi zram-config[520]: + read -r line
Feb 08 02:26:37 raspberrypi zram-config[520]: + case "$line" in
Feb 08 02:26:37 raspberrypi zram-config[520]: + continue
Feb 08 02:26:37 raspberrypi zram-config[520]: + read -r line
Feb 08 02:26:37 raspberrypi zram-config[520]: + case "$line" in
Feb 08 02:26:37 raspberrypi zram-config[520]: + continue
Feb 08 02:26:37 raspberrypi zram-config[520]: + read -r line
Feb 08 02:26:37 raspberrypi zram-config[520]: + case "$line" in
Feb 08 02:26:37 raspberrypi zram-config[520]: + continue
Feb 08 02:26:37 raspberrypi zram-config[520]: + read -r line
Feb 08 02:26:37 raspberrypi zram-config[520]: + case "$line" in
Feb 08 02:26:37 raspberrypi zram-config[520]: + continue
Feb 08 02:26:37 raspberrypi zram-config[520]: + read -r line
Feb 08 02:26:37 raspberrypi zram-config[520]: + case "$line" in
Feb 08 02:26:37 raspberrypi zram-config[520]: + continue
Feb 08 02:26:37 raspberrypi zram-config[520]: + read -r line
Feb 08 02:26:37 raspberrypi zram-config[520]: + case "$line" in
Feb 08 02:26:37 raspberrypi zram-config[520]: + continue
Feb 08 02:26:37 raspberrypi zram-config[520]: + read -r line
Feb 08 02:26:37 raspberrypi zram-config[520]: + case "$line" in
Feb 08 02:26:37 raspberrypi zram-config[520]: + continue
Feb 08 02:26:37 raspberrypi zram-config[520]: + read -r line
Feb 08 02:26:37 raspberrypi zram-config[520]: + case "$line" in
Feb 08 02:26:37 raspberrypi zram-config[520]: + continue
Feb 08 02:26:37 raspberrypi zram-config[520]: + read -r line
Feb 08 02:26:37 raspberrypi zram-config[520]: + case "$line" in
Feb 08 02:26:37 raspberrypi zram-config[520]: + continue
Feb 08 02:26:37 raspberrypi zram-config[520]: + read -r line
Feb 08 02:26:37 raspberrypi zram-config[520]: + case "$line" in
Feb 08 02:26:37 raspberrypi zram-config[520]: + continue
Feb 08 02:26:37 raspberrypi zram-config[520]: + read -r line
Feb 08 02:26:37 raspberrypi zram-config[520]: + case "$line" in
Feb 08 02:26:37 raspberrypi zram-config[520]: + continue
Feb 08 02:26:37 raspberrypi zram-config[520]: + read -r line
Feb 08 02:26:37 raspberrypi zram-config[520]: + case "$line" in
Feb 08 02:26:37 raspberrypi zram-config[520]: + continue
Feb 08 02:26:37 raspberrypi zram-config[520]: + read -r line
Feb 08 02:26:37 raspberrypi zram-config[520]: + case "$line" in
Feb 08 02:26:37 raspberrypi zram-config[520]: + continue
Feb 08 02:26:37 raspberrypi zram-config[520]: + read -r line
Feb 08 02:26:37 raspberrypi zram-config[520]: + case "$line" in
Feb 08 02:26:37 raspberrypi zram-config[520]: + continue
Feb 08 02:26:37 raspberrypi zram-config[520]: + read -r line
Feb 08 02:26:37 raspberrypi zram-config[520]: + case "$line" in
Feb 08 02:26:37 raspberrypi zram-config[520]: + continue
Feb 08 02:26:37 raspberrypi zram-config[520]: + read -r line
Feb 08 02:26:37 raspberrypi zram-config[520]: + case "$line" in
Feb 08 02:26:37 raspberrypi zram-config[520]: + continue
Feb 08 02:26:37 raspberrypi zram-config[520]: + read -r line
[... 145 lines cut ...]
Feb 08 02:26:38 raspberrypi zram-config[520]: + OLDLOG_DIR=/opt/zram/oldlog
Feb 08 02:26:38 raspberrypi zram-config[520]: + serviceConfiguration stop
Feb 08 02:26:38 raspberrypi zram-config[520]: + [[ stop == \s\t\o\p ]]
Feb 08 02:26:38 raspberrypi zram-config[520]: + echo 'Stopping services that interfere with zram device configuration'
Feb 08 02:26:38 raspberrypi zram-config[520]: ++ systemctl is-active rsyslog.service
Feb 08 02:26:39 raspberrypi zram-config[520]: + [[ inactive == \a\c\t\i\v\e ]]
Feb 08 02:26:39 raspberrypi zram-config[520]: ++ systemctl is-active systemd-journald.service
Feb 08 02:26:39 raspberrypi zram-config[520]: + [[ active == \a\c\t\i\v\e ]]
Feb 08 02:26:39 raspberrypi zram-config[520]: + export journaldActive=true
Feb 08 02:26:39 raspberrypi zram-config[520]: + journaldActive=true
Feb 08 02:26:39 raspberrypi zram-config[520]: + journalctl --flush
Feb 08 02:26:39 raspberrypi zram-config[520]: + systemctl stop systemd-journald.socket systemd-journald-audit.socket systemd-journald-dev-log.socket
[... cut ...]
Feb 08 02:26:42 raspberrypi kernel: zram: Added device: zram1
Feb 08 02:26:42 raspberrypi systemd[1208]: zram-setup@zram1.service: Failed to connect stdout to the journal socket, ignoring: Connection refused
Feb 08 02:26:42 raspberrypi systemd[1209]: zram-setup@zram1.service: Failed to connect stdout to the journal socket, ignoring: Connection refused
Feb 08 02:26:42 raspberrypi systemd[1211]: zram-setup@zram1.service: Failed to connect stdout to the journal socket, ignoring: Connection refused

yutayu, 2021-02-07:
pi@raspberrypi:~ $ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root       916G  797G   74G  92% /
devtmpfs        430M     0  430M   0% /dev
tmpfs           463M   51M  412M  11% /dev/shm
tmpfs           463M   13M  450M   3% /run
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           463M     0  463M   0% /sys/fs/cgroup
tmpfs           463M  412K  462M   1% /tmp
/dev/mmcblk0p1  253M   48M  205M  19% /boot
tmpfs            93M  8.0K   93M   1% /run/user/1000
pi@raspberrypi:~ $ zramctl
NAME       ALGORITHM DISKSIZE DATA COMPR TOTAL STREAMS MOUNTPOINT
/dev/zram0 lzo-rle       250M   4K   86B   12K       4 [SWAP]

yutayu, 2021-02-07:
# swap	alg		mem_limit	disk_size	swap_priority	page-cluster	swappiness
swap	lzo-rle		250M		750M		75		0		80
#swap	lzo-rle		400M		1200M		75		0		80

# dir	alg		mem_limit	disk_size	target_dir		bind_dir
#dir	lzo-rle		50M		150M		/home/pi		/pi.bind
#dir	lzo-rle		100M		300M		/home/pi		/pi.bind

# log	alg		mem_limit	disk_size	target_dir		bind_dir		oldlog_dir
log	lzo-rle		50M		150M		/var/log		/log.bind		/opt/zram/oldlog
#log	lzo-rle		50M		150M		/var/log		/log.bind		/opt/zram/oldlog

ecdye, 2021-02-07:
This tells me nothing, what is your issue zram-config seems to be working from the output you provided. Please describe what your issue is not just make blanket statements about it not working.

yutayu, 2021-02-07:
I wanted to set swap size 750M, but at last 250M. 
and /var/log is not on zram.

ecdye, 2021-02-07:
It seems that it is working fine for the swap. Have you tried to reboot before starting zram for the log? Additionally make sure that if there are services accessing the log directory that they are stopped before starting zram-config. I have tested this on my systems and never observed an issue.

yutayu, 2021-02-07:
In this case , 250M on ztab is as memlimit of swap.
and I rebooted pi again , it re-produces.

Is your situtation is no problem ? strange.

ecdye, 2021-02-07:
Yes that is correct. I don't see what your issue is.

yutayu, 2021-02-07:
swap	alg		mem_limit	disk_size	swap_priority	page-cluster	swappiness
swap lzo-rle 250M 750M 75 0 80

I set swapsize 750M as disk_size,
but zram-config made 250M.

This is problem.

ecdye, 2021-02-07:
250 is the memory limit, 750 is the disk limit. The maximum data allowed in uncompressed size will be 750M, but the zram will limit the memory or RAM usage of this to 250M which is about right for the compression algorithm used. It is working properly. 

Please read the documentation at <https://github.com/ecdye/zram-config#configure>

closed 2021-02-07 by ecdye

yutayu, 2021-02-07:
see , zram-config made 250M as DISKSIZE.
so 
It should be 750M as DISKSIZE.

yutayu, 2021-02-07:
at ztab
swap lzo-rle 250M 750M 75 0 80
at ztamctl
/dev/zram0 lzo-rle 250M 4K 86B 12K 4 [SWAP]

This is not correct result.

ecdye, 2021-02-07:
Thank you for finally telling me the issue, it is now fixed.

yutayu, 2021-02-07:
Thank you for fixing , DISKSIZE is correct.

next is disk number.
zram-config made 1 directories.
It should be 3 directories.

yutayu, 2021-02-07:
Second case
at ztab
swap	lzo-rle		400M		1600M		75		0		80
dir	lzo-rle		50M		200M		/home/pi		/pi.bind
log	lzo-rle		12M		48M		/var/log		/log.bind		/opt/zram/oldlog

at  zramctl
NAME       ALGORITHM DISKSIZE  DATA COMPR TOTAL STREAMS MOUNTPOINT
/dev/zram1 lzo-rle        48M  4.3M 32.6K  264K       4 /opt/zram/zram1
/dev/zram0 lzo-rle       1.6G    4K   87B   12K       4 [SWAP]

it should be 3 zrams.

yutayu, 2021-02-07:
[zram.txt](https://github.com/ecdye/zram-config/files/5939838/zram.txt)
journalctl -xe | grep zram-config > zram.txt

ecdye, 2021-02-07:
Not helpful, please provide the logs located at `/usr/local/share/zram-config/log/zram-config.log`

yutayu, 2021-02-07:
[zram-config.log](https://github.com/ecdye/zram-config/files/5939953/zram-config.log)

ecdye, 2021-02-07:
Thank you, you appear to only have the `log` and `swap` options enabled. 

Please give the output of running the following:

``` bash
cat /etc/ztab
```

yutayu, 2021-02-07:
[zram2.txt](https://github.com/ecdye/zram-config/files/5940058/zram2.txt)

ecdye, 2021-02-09:
My best guess is you might be using more RAM then you have as I have no issues with creating all the devices on my side. This is not an issue with the code and as such I am not willing to debug it any further.

yutayu, 2021-02-09:
ok

commit b18b6e84 (2021-02-07): Set disk size properly when creating zram device

Fixes #58