ecdye/zram-config #60: missing dependencies

bernstein82, 2021-02-15:
on proxmox ve 6.3 (debian buster) following the install instructions didn't work:
```
~/zram-config# ./install.bash
Installing needed packages (make, libattr1-dev)
gcc -Wall -std=c99 -c main.c
make: gcc: Command not found
make: *** [makefile:11: main.o] Error 127
Installing zram-config files
install: cannot stat 'overlayfs-tools/overlay': No such file or directory
Starting zram-config.service
Created symlink /etc/systemd/system/basic.target.wants/zram-config.service → /etc/systemd/system/zram-config.service.
#####   zram-config is now installed and running   #####
#####     edit /etc/ztab to configure options      #####
```
had to manually `apt install make gcc`, to make it work.

ecdye, 2021-02-15:
It is assumed that on a regular Debian installation, the `build-essential` package is installed. I will add it to the list of required packages for the edge case scenario that it is not.

closed 2021-02-15 by ecdye

reukiodo, 2025-02-21:
Why is the entire build system needed just for installation? Is there a .deb which can have a pre-compiled version?

ecdye, 2025-02-21:
As of right now, no there is not a precompiled deb package that can be used. At some point I would like create one, but it is not very high on the priority list right now for a variety of reasons.

commit 0669f656 (2021-02-15): Require build-essential for install process

Requiring build-essential ensures that all packages needed to build the
overlayfs-tools project are installed and available during the execution
of the install script. Before when only requiring make other needed
packages like gcc could not be present at build time resulting in a
failed install.

Fixes #60