ecdye/zram-config #63: Lots of output after sudo zram-config "stop" / "start"

RURon, 2021-04-14:
I have lots of shell output that looks like unwanted behaviour and so I wonder if this is a problem related to my system or not. I'm running Debian Buster on a RaspberryPi4.

```
[09:07:12] pi@raspberrypi4:~$ sudo zram-config "stop"
+ TMPDIR=/tmp
+ ZSHARE=/usr/local/share/zram-config
+ ZLOG=/usr/local/share/zram-config/log/zram-config.log
+ ZDIR=/opt/zram
+ [[ '' == \s\h\u\t\d\o\w\n ]]
+ case "$1" in
++ systemctl is-active zram-config.service
+ [[ active == \a\c\t\i\v\e ]]
+ systemctl kill zram-config.service
+ systemctl stop zram-config.service
++ date +%Y-%m-%d-%H:%M:%S
+ echo 'zram-config stop 2021-04-14-09:07:17'
+ tac /tmp/zram-device-list
+ file=/tmp/zram-device-list.rev
+ read -r line
+ case "$line" in
+ set -- log /zram2 /var/log /log.bind
+ echo 'ztab remove log /zram2 /var/log /log.bind'
+ case "$1" in
+ ZRAM_DEV=/zram2
+ TARGET_DIR=/var/log
+ BIND_DIR=/log.bind
+ [[ -z '' ]]
+ serviceConfiguration stop
+ [[ stop == \s\t\o\p ]]
+ echo 'Stopping services that interfere with zram device configuration'
++ systemctl is-active rsyslog.service
+ [[ active == \a\c\t\i\v\e ]]
+ export rsyslogActive=true
+ rsyslogActive=true
+ systemctl stop syslog.socket
++ systemctl is-active systemd-journald.service
+ [[ active == \a\c\t\i\v\e ]]
+ export journaldActive=true
+ journaldActive=true
+ journalctl --flush
+ systemctl stop systemd-journald.socket systemd-journald-audit.socket systemd-journald-dev-log.socket
+ removeZdir
+ [[ -n '' ]]
+ echo /zram2
+ [[ -n /var/log ]]
+ umount --verbose /var/log/
+ umount --verbose --lazy /var/log/
+ return 1
+ read -r line
+ case "$line" in
+ set -- swap /zram0 zram-config0
+ echo 'ztab remove swap /zram0 zram-config0 '
+ case "$1" in
+ ZRAM_DEV=/zram0
+ removeZswap
+ swapoff /dev/zram0
+ return 1
+ read -r line
+ rm -fv /tmp/zram-device-list.rev /tmp/zram-device-list
+ [[ -z '' ]]
+ serviceConfiguration start
+ [[ start == \s\t\o\p ]]
+ [[ start == \s\t\a\r\t ]]
+ echo 'Restarting services that interfere with zram device configuration'
+ [[ -n true ]]
+ systemctl restart systemd-journald.socket systemd-journald-audit.socket systemd-journald-dev-log.socket
+ [[ -n true ]]
+ systemctl restart syslog.socket
```

ecdye, 2021-04-14:
Nope, that is all just the process of zram-config stopping the zram process. That is all normal nothing unusual there.

RURon, 2021-04-14:
Ok, this is pretty verbose for just a service being stopped (or started) - which I find okay while debugging but find too much information when running in a regular system (I need to stop zram before I run a backup and get all the output in an email, which clutters up the email with information that I dont really need). Is there nothing I can do to make it less verbose? (other than completely ditching all output to /dev/null)

ecdye, 2021-04-14:
Currently no, there is nothing implemented to make it stop. It does log everything to a file so even if you do silence the output it should be fine.

ecdye, 2021-04-18:
After further thought, I am going to remove the verbose logging as in most cases it is unnecessary and instead will output just some simple statements to indicate what is happening to `stdout` and then log sightly more in the log file.

commit 5c965716 (2021-04-18): Lower normal logging verbosity

Fixes #63
closed 2021-04-18 by ecdye