ecdye/zram-config #67: Not mounting any target

domy-os, 2021-05-29:
Hi guys,

I do have Raspberry PI 4 8 GB with fully updated Ubuntu Mate 20.04.2 LTS. I'm trying to set this thing up but maybe I'm stupid because it's not working like you have presented on the code home page.

`domy@domy-pi:~$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.2 LTS
Release:        20.04
Codename:       focal`

`domy@domy-pi:~$ zramctl
domy@domy-pi:~$`

`domy@domy-pi:~$ systemctl status zram-config.service
● zram-config.service - zram-config
     Loaded: loaded (/etc/systemd/system/zram-config.service; enabled; vendor preset: enabled)
     Active: active (exited) since Sat 2021-05-29 20:32:33 CEST; 10min ago
       Docs: https://github.com/ecdye/zram-config/blob/main/README.md
    Process: 1918 ExecStart=/usr/local/sbin/zram-config start (code=exited, status=0/SUCCESS)
   Main PID: 1918 (code=exited, status=0/SUCCESS)

May 29 20:32:33 domy-pi systemd[1]: Started zram-config.
May 29 20:32:33 domy-pi zram-config[1923]: zram-config start 2021-05-29-20:32:33
May 29 20:32:34 domy-pi zram-config[1961]: zramctl: /dev/zram0: failed to reset: Device or resource busy`

`domy@domy-pi:~$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
loop0         7:0    0 127.1M  1 loop /snap/chromium/1595
loop1         7:1    0  48.9M  1 loop /snap/core18/1949
loop2         7:2    0 145.4M  1 loop /snap/gnome-3-28-1804/147
loop3         7:3    0  65.1M  1 loop /snap/gtk-common-themes/1515
loop4         7:4    0  48.9M  1 loop /snap/core18/2068
loop5         7:5    0  59.6M  1 loop /snap/lxd/20330
loop6         7:6    0  61.6M  1 loop /snap/lxd/19040
loop7         7:7    0  27.9M  1 loop /snap/snapd/11843
loop8         7:8    0    27M  1 loop /snap/snapd/10709
loop9         7:9    0    16K  1 loop /snap/software-boutique/56
loop10        7:10   0  15.2M  1 loop /snap/ubuntu-mate-welcome/627
mmcblk0     179:0    0 119.1G  0 disk
|-mmcblk0p1 179:1    0   256M  0 part /boot/firmware
`-mmcblk0p2 179:2    0 118.9G  0 part /`

`domy@domy-pi:~$ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            3.7G     0  3.7G   0% /dev
tmpfs           782M   30M  752M   4% /run
/dev/mmcblk0p2  117G  6.4G  106G   6% /
tmpfs           3.9G     0  3.9G   0% /dev/shm
tmpfs           5.0M  4.0K  5.0M   1% /run/lock
tmpfs           3.9G     0  3.9G   0% /sys/fs/cgroup
/dev/loop0      128M  128M     0 100% /snap/chromium/1595
/dev/loop1       49M   49M     0 100% /snap/core18/1949
/dev/loop2      146M  146M     0 100% /snap/gnome-3-28-1804/147
/dev/loop3       66M   66M     0 100% /snap/gtk-common-themes/1515
/dev/loop4       49M   49M     0 100% /snap/core18/2068
/dev/loop9      128K  128K     0 100% /snap/software-boutique/56
/dev/loop6       62M   62M     0 100% /snap/lxd/19040
/dev/loop7       28M   28M     0 100% /snap/snapd/11843
/dev/loop5       60M   60M     0 100% /snap/lxd/20330
/dev/loop8       27M   27M     0 100% /snap/snapd/10709
/dev/loop10      16M   16M     0 100% /snap/ubuntu-mate-welcome/627
/dev/mmcblk0p1  253M  128M  125M  51% /boot/firmware
tmpfs           782M   16K  782M   1% /run/user/1000`

`domy@domy-pi:~$ swapon -s
domy@domy-pi:~$`

Log file is attached, a little bit long for this. [zram-config.log](https://github.com/ecdye/zram-config/files/6565195/zram-config.log)
Configuration file /etc/ztab has not been changed yet.

Can someone please explain me how to get this thing working? And is it possible to include /tmp directory with this tool or I do need to use /etc/fstab?

Thank you in advance and I wish you all the best!

ecdye, 2021-05-29:
Alright, I am not surprised that you have had issues with this. First off, RPi 4 8GB does not always work correctly with zram right now. This is due to a Linux Kernel issue that is ongoing. While not a part of this project, [this](https://community.openhab.org/t/110703) post on the openHAB forum has some details. For the moment there are no real good ways to enable zram on a RPi 4 8GB it may be possible but I won't try and support it as the methods used would be very hacky.

Second, it should not be necessary to put the `/tmp` directory on zram please read the [Performance](https://github.com/ecdye/zram-config#performance) section of the README to get more details.

I will make sure to make a note in the README so that people know that the RPi 8GB is not currently working with zram due to this issue.

domy-os, 2021-05-29:
> Alright, I am not surprised that you have had issues with this. First off, RPi 4 8GB does not always work correctly with zram right now. This is due to a Linux Kernel issue that is ongoing. While [...]

Aaahh, I see, I didn't even think about that, lol.

> Second, it should not be necessary to put the `/tmp` directory on zram please read the [Performance](https://github.com/ecdye/zram-config#performance) section of the README to get more details.

I saw this section but my /tmp is not mounted in RAM but in the root.

```
domy@domy-pi:~$ df -h /tmp
Filesystem      Size  Used Avail Use% Mounted on
/dev/mmcblk0p2  117G  6.4G  106G   6% /
domy@domy-pi:~$
```

> I will make sure to make a note in the README so that people know that the RPi 8GB is not currently working with zram due to this issue.

Thanks for your time!

Anyhow, I do have around 5 MB writes per day, it's not that urgent, I'll wait for a fix.

ecdye, 2021-05-29:
Ahh, well then you would likely be better suited with using `/etc/fstab` to mount your `/tmp` directory on ram rather than zram.

domy-os, 2021-05-30:
Actually, I've found that tmp.mount is not enabled by default, we need to copy *.mount file and enable it.

```
sudo cp -v /usr/share/systemd/tmp.mount /etc/systemd/system/ 
sudo systemctl enable tmp.mount
```

So, I will close this issue, there is nothing we can do right now.

Best regards!

closed 2021-05-30 by domy-os