ecdye/zram-config #69: Difference to openhabian-zram

vendetta1987, 2021-08-03:
I'm asking myself what might be the difference between this project and [openhabian-zram](https://github.com/mstormi/openhabian-zram). As you're contributing to both and they seem to do the same thing this might be of interest.
I was using your project on my rpi4 with OpenHab running inside a docker container. Recently I ran into the problem of the 'zram-ed' files and folders not beeing synchronized to disk during reboots. As openhabian-zram obviously targets OpenHab I just gave it a try now.
Would you be willing to elaborate on the differences apart from the ephemeral mode?

ecdye, 2021-08-07:
Good question! openhabian-zram was a fork of this project the openHABian maintainers had created to allow for openHAB specific changes to be made to the codebase. I later ended up taking over maintainership for this project and was then able to simply create different git branches to make those openHAB specific changes. This is the most up to date version of zram-config and should be used preferentially as there have been many bug fixes that have been made in the last 6 months.

Running openHAB inside of a docker container on a RPi is not a supported configuration for zram and openHAB, I would recommend trying openHABian instead to allow for much more support and configuration. Not to mention it does its best to set up all of these things automatically so the user does not need to worry.

Ephemeral mode was removed because of the GPLv3's strict licensing requirements that this projects original license did not meet. More details on that change can be found in this commit that removed it <https://github.com/ecdye/zram-config/commit/9693730ced4a93d6be5b3fa8b814f375a539c7d6>.

Please let me know if this helps and if you have any further questions.

vendetta1987, 2021-08-10:
Good to know, thank you. Maybe the two projects can be merged or one based on the other in the future.
I'm going to check my setup, maybe my problems are based on systemd service execution order.

closed 2021-08-10 by vendetta1987