ecdye/zram-config #70: Service is active, but no zrams are mounted

littlemax28, 2021-08-06:
Fresh install of debian 10 with pihole.
I've just installed the zram-config. Used the default /etc/ztab-file. Service is active, but zramctl shows empty output. DF, FREE -H and SWAPON commands shows no signs of any zram drives mounted.

Stop command finished with error (see - below).
root@wyse:/opt/zram# sudo zram-config "stop"
zram-config stop 2021-08-06-23:24:51
tac: failed to open '/tmp/zram-device-list' for reading: No such file or directory

I would be gratefull for any suggestions how to make it working.

Linux wyse 4.19.0-17-amd64 #1 SMP Debian 4.19.194-3 (2021-07-18) x86_64 GNU/Linux

littlemax28, 2021-08-07:
Checked the status of service (see below) and noticed that the error at the last line "failed to set alg".
Checked the /proc/crypto file and found no lzo-rle.
I guess zlo was't installed. I install zram-tools and after that everything started to work. 
I assume you should add certain dependencies or something like that to zram-config's install script.

```
root@wyse:~/zram-config# sudo systemctl status zram-config.service
● zram-config.service - zram-config
   Loaded: loaded (/etc/systemd/system/zram-config.service; enabled; vendor p
   Active: active (exited) since Sat 2021-08-07 19:28:13 MSK; 15s ago
     Docs: https://github.com/ecdye/zram-config/blob/main/README.md
  Process: 710 ExecStart=/usr/local/sbin/zram-config start (code=exited, stat
 Main PID: 710 (code=exited, status=0/SUCCESS)

Aug 07 19:28:13 wyse systemd[1]: Started zram-config.
Aug 07 19:28:13 wyse zram-config[710]: zram-config start 2021-08-07-19:28:13
Aug 07 19:28:14 wyse zram-config[710]: zramctl: /dev/zram0: failed to set alg
```

littlemax28, 2021-08-07:
I was too fast to close this issue.
So far after installing zram-tools, stop and start zram-config.service I got zswap, but not log and dir zrams mounted.
I rebooted the system but now luck.
Here is my /etc/ztab 
```
# swap  alg             mem_limit       disk_size       swap_priority   page$
swap    lzo-rle         250M            750M            75              0   $

# dir   alg             mem_limit       disk_size       target_dir          $
dir     lzo-rle         150M            450M            /var/cache          $

# log   alg             mem_limit       disk_size       target_dir          $
log     lzo-rle         250M            750M            /var/log            $

```

and some command's output:
```
root@wyse:~# sudo zramctl
NAME       ALGORITHM DISKSIZE DATA COMPR TOTAL STREAMS MOUNTPOINT
/dev/zram3 lzo            64M   4K   78B   12K       4 [SWAP]
/dev/zram2 lzo            64M   4K   78B   12K       4 [SWAP]
/dev/zram0 lzo            64M   4K   78B   12K       4 [SWAP]
```
after reboot I found zram2 and zram3 in addition to zram0 which was there before reboot

```
root@wyse:~# df
Filesystem     1K-blocks    Used Available Use% Mounted on
udev              961192       0    961192   0% /dev
tmpfs             195216    3164    192052   2% /run
/dev/mmcblk0p2   5947704 1639748   3986116  30% /
tmpfs             976072    2320    973752   1% /dev/shm
tmpfs               5120       0      5120   0% /run/lock
tmpfs             976072       0    976072   0% /sys/fs/cgroup
/dev/mmcblk0p1    523248    6704    516544   2% /boot/efi
tmpfs             195212       0    195212   0% /run/user/998
tmpfs             195212       0    195212   0% /run/user/1000
root@wyse:~# free -h
              total        used        free      shared  buff/cache   available
Mem:          1.9Gi       123Mi       1.3Gi       8.0Mi       438Mi       1.6Gi
Swap:         1.1Gi          0B       1.1Gi
root@wyse:~# sudo swapon
NAME           TYPE      SIZE USED PRIO
/dev/mmcblk0p3 partition 976M   0B   -2
/dev/zram0     partition  64M   0B  100
/dev/zram2     partition  64M   0B  100
/dev/zram3     partition  64M   0B  100

```
still need help

littlemax28, 2021-08-07:
```
root@wyse:/opt/zram# journalctl -xe | grep zram
Aug 07 20:16:01 wyse systemd[1]: Starting Linux zramswap setup...
-- Subject: A start job for unit zramswap.service has begun execution
-- A start job for unit zramswap.service has begun execution.
Aug 07 20:16:01 wyse systemd[1]: Started zram-config.
-- Subject: A start job for unit zram-config.service has finished successfully
-- A start job for unit zram-config.service has finished successfully.
Aug 07 20:16:01 wyse zram-config[377]: zram-config start 2021-08-07-20:16:01
Aug 07 20:16:01 wyse kernel: zram: Added device: zram0
Aug 07 20:16:01 wyse kernel: zram: Added device: zram1
Aug 07 20:16:01 wyse kernel: zram: Added device: zram2
Aug 07 20:16:01 wyse kernel: zram: Added device: zram3
Aug 07 20:16:01 wyse kernel: zram0: detected capacity change from 0 to 67108864
Aug 07 20:16:01 wyse zramswap[376]: Setting up swapspace version 1, size = 64 MiB (67104768 bytes)
Aug 07 20:16:01 wyse zramswap[376]: no label, UUID=7e70647a-3777-48f7-9790-1fa449ed9539
Aug 07 20:16:01 wyse kernel: Adding 65532k swap on /dev/zram0.  Priority:100 extents:1 across:65532k SSFS
Aug 07 20:16:01 wyse kernel: zram1: detected capacity change from 0 to 67108864
Aug 07 20:16:01 wyse kernel: zram1: detected capacity change from 67108864 to 0
Aug 07 20:16:01 wyse zramswap[376]: mkswap: error: swap area needs to be at least 40 KiB
Aug 07 20:16:01 wyse zram-config[377]: zramctl: /dev/zram1: failed to set algorithm: Invalid argument
Aug 07 20:16:01 wyse zramswap[376]: swapon: /dev/zram1: read swap header failed
Aug 07 20:16:01 wyse kernel: zram2: detected capacity change from 0 to 67108864
Aug 07 20:16:01 wyse zramswap[376]: Setting up swapspace version 1, size = 64 MiB (67104768 bytes)
Aug 07 20:16:01 wyse zramswap[376]: no label, UUID=9089f825-94b9-4e89-a24c-1c21cc912c62
Aug 07 20:16:01 wyse kernel: Adding 65532k swap on /dev/zram2.  Priority:100 extents:1 across:65532k SSFS
Aug 07 20:16:01 wyse kernel: zram3: detected capacity change from 0 to 67108864
Aug 07 20:16:01 wyse zramswap[376]: Setting up swapspace version 1, size = 64 MiB (67104768 bytes)
Aug 07 20:16:01 wyse zramswap[376]: no label, UUID=a02f38ec-1095-4a5c-beba-8a9f645f8d9e
Aug 07 20:16:01 wyse kernel: Adding 65532k swap on /dev/zram3.  Priority:100 extents:1 across:65532k SSFS
Aug 07 20:16:01 wyse systemd[1]: Started Linux zramswap setup.
-- Subject: A start job for unit zramswap.service has finished successfully
-- A start job for unit zramswap.service has finished successfully.
Aug 07 20:16:47 wyse sudo[624]:     root : TTY=pts/0 ; PWD=/home/user ; USER=root ; COMMAND=/usr/sbin/zramctl
Aug 07 20:24:29 wyse sudo[4249]:     root : TTY=pts/0 ; PWD=/home/user ; USER=root ; COMMAND=/usr/local/sbin/zram-config stop
Aug 07 20:24:29 wyse systemd[1]: Stopping zram-config...
-- Subject: A stop job for unit zram-config.service has begun execution
-- A stop job for unit zram-config.service has begun execution.
Aug 07 20:24:29 wyse zram-config[4257]: zram-config stop 2021-08-07-20:24:29
Aug 07 20:24:29 wyse zram-config[4257]: tac: failed to open '/tmp/zram-device-list' for reading: No such file or directory
Aug 07 20:24:29 wyse systemd[1]: zram-config.service: Succeeded.
-- The unit zram-config.service has successfully entered the 'dead' state.
Aug 07 20:24:29 wyse systemd[1]: Stopped zram-config.
-- Subject: A stop job for unit zram-config.service has finished
-- A stop job for unit zram-config.service has finished.
Aug 07 20:24:42 wyse sudo[4267]:     root : TTY=pts/0 ; PWD=/home/user ; USER=root ; COMMAND=/usr/bin/systemctl status zram-config.service
Aug 07 20:24:54 wyse sudo[4270]:     root : TTY=pts/0 ; PWD=/home/user ; USER=root ; COMMAND=/usr/bin/systemctl start zram-config.service
Aug 07 20:24:54 wyse systemd[1]: Started zram-config.
-- Subject: A start job for unit zram-config.service has finished successfully
-- A start job for unit zram-config.service has finished successfully.
Aug 07 20:24:54 wyse zram-config[4273]: zram-config start 2021-08-07-20:24:54
Aug 07 20:24:54 wyse zram-config[4273]: zramctl: /dev/zram1: failed to set algorithm: Invalid argument
Aug 07 20:24:58 wyse sudo[4333]:     root : TTY=pts/0 ; PWD=/home/user ; USER=root ; COMMAND=/usr/bin/systemctl status zram-config.service
Aug 07 20:26:15 wyse sudo[4353]:     root : TTY=pts/0 ; PWD=/home/user ; USER=root ; COMMAND=/usr/sbin/zramctl
Aug 07 20:45:22 wyse sudo[4980]:     root : TTY=pts/0 ; PWD=/root ; USER=root ; COMMAND=/usr/sbin/zramctl
Aug 07 21:03:33 wyse sudo[6467]:     root : TTY=pts/1 ; PWD=/opt/zram ; USER=root ; COMMAND=/usr/local/sbin/zram-config stop
Aug 07 21:03:34 wyse systemd[1]: Stopping zram-config...
-- Subject: A stop job for unit zram-config.service has begun execution
-- A stop job for unit zram-config.service has begun execution.
Aug 07 21:03:34 wyse zram-config[6475]: zram-config stop 2021-08-07-21:03:34
Aug 07 21:03:34 wyse zram-config[6475]: tac: failed to open '/tmp/zram-device-list' for reading: No such file or directory
Aug 07 21:03:34 wyse systemd[1]: zram-config.service: Succeeded.
-- The unit zram-config.service has successfully entered the 'dead' state.
Aug 07 21:03:34 wyse systemd[1]: Stopped zram-config.
-- Subject: A stop job for unit zram-config.service has finished
-- A stop job for unit zram-config.service has finished.
Aug 07 21:04:50 wyse systemd[1]: Started zram-config.
-- Subject: A start job for unit zram-config.service has finished successfully
[... cut ...]
-- Subject: A stop job for unit zram-config.service has finished
-- A stop job for unit zram-config.service has finished.
Aug 07 21:08:12 wyse systemd[1]: Started zram-config.
-- Subject: A start job for unit zram-config.service has finished successfully
-- A start job for unit zram-config.service has finished successfully.
Aug 07 21:08:12 wyse zram-config[6845]: zram-config start 2021-08-07-21:08:12
Aug 07 21:08:12 wyse zram-config[6845]: zramctl: /dev/zram1: failed to set algorithm: Invalid argument

```

littlemax28, 2021-08-07:
[zram-config_log.txt](https://github.com/ecdye/zram-config/files/6949594/zram-config_log.txt)

ecdye, 2021-08-07:
Simple solution, your linux kernel is too old and does not support `lzo-rle`. Try changing your algorithm in `/etc/ztab` to `lz4` and see if it works.

littlemax28, 2021-08-08:
The change of algorithm did the job.
Thanks a lot.

closed 2021-08-08 by littlemax28