ecdye/zram-config #72: Logs are written to HDD only on shutdown

Jack20947, 2021-11-29:
At the moment zram-config merges the content of the upper overlay in RAM to the hdd (or SD card) only if the service is stopped manually or during shutdown.
There may be many use-case where the machine is not restarted for many days, for example  a raspberry pi that is used for pihole, vpn, and so on. And in this case, if a power interruption happens, the log is lost.
Is it possible to add a functionality that copy the RAM overlay to the hdd periodically? (once a day for example)

ecdye, 2021-11-29:
Possible too, yes, it is certainly a feature that could be useful. I would like to remind you that the real solution to this problem is to purchase a UPS and use that if the data you are logging or putting in zram is important enough to warrant the need of the daily backup solution.

ecdye, 2021-12-02:
After a closer look at the code, this would likely require a restart of zram to accomplish the desired outcome of syncing the data to the disk. This is because of the way the overlayfs operates and when syncing it is ideal to unmount the overlayed directory necessitating a stop of the zram process. 

As a result of this, my current recommendation until such a time that there have been real official tools created for overlayfs (unlikely) is to use a service like cron to restart zram-config as often as you would like.

Jack20947, 2021-12-02:
Ok thanks. Also https://github.com/azlux/log2ram uses a cron to rsync the content of the ram to the disk. so I suppose it can be done without restarting the service. (I'm not using ram2log because I can't use the zram as swap with it).

ecdye, 2021-12-02:
LogToRam can do that because it does not use the overlayfs system that zram-config uses. As I mentioned earlier, the overlayfs is what really prevents me from doing the same thing in zram-config.

ecdye, 2022-02-07:
There may actually be a way to do this after some research. I will report back with my findings.

PR #79 (2022-02-07): Allow sync to persistent storage
Using the same merge functionality in overlayfs-tools already used to
write changes to persistent storage it is possible to write changes to
disk without completely reseting the zram device. This allows for a
quick sync to persistent storage to be run whenever desired.

Without the ability to sync, any changes made to the underlying
filesystem will be lost if there was a sudden power loss or system
shutdown. This can be mitigated by periodically running the sync command
which will write the changes to persistent storage ensuring that the
data will not be lost i [... cut]
commit 1a028f78 (2022-02-07): Allow sync to persistent storage (#79)

Using the same merge functionality in overlayfs-tools already used to
write changes to persistent storage it is possible to write changes to
disk without completely reseting the zram device. This allows for a
quick sync to persistent storage to be run whenever desired.

Without the ability to sync, any changes made to the underlying
filesystem will be lost if there was a sudden power loss or system
shutdown. This can be mitigated by periodically running the sync command
which will write the changes to persistent storage ensuring that the
data will not be [... cut]
closed 2022-02-07 by ecdye