ecdye/zram-config #85: gpg Can't check signature: No public key A2A3BF1A7D8D7AD47E939274B13E6AF0E31D89F8

MaxPeal, 2022-03-08:
the gpg Signature public key is missing.

MaxPeal, 2022-03-08:
```
#!/bin/bash
set -Eeuo pipefail
#set -u pipefail
export PROG_KEYS=A2A3BF1A7D8D7AD47E939274B13E6AF0E31D89F8
export PROG_VERSION=1.4.0
export PROG_URL=https://github.com/ecdye/zram-config/releases/download/v"$PROG_VERSION"/zram-config-v"$PROG_VERSION".tar.lz
export PROG_SRC_DIR=/usr/src/zram-config
	tarball="$(basename "$PROG_URL")"; \
	wget -O "$tarball.asc" "$PROG_URL.asc"; \
	wget -O "$tarball" "$PROG_URL" --progress=dot:giga; \
	\
	export GNUPGHOME="$(mktemp -d)"; \
	for key in $PROG_KEYS; do \
		gpg --batch --keyserver keyserver.ubuntu.com --recv-keys "$key"; \
	done; \
	gpg --batch --verify "$tarball.asc" "$tarball"; \
	gpgconf --kill all; \
	rm -rf "$GNUPGHOME"; \
	\
	mkdir "$PROG_SRC_DIR"; \
	tar -xf "$tarball" -C "$PROG_SRC_DIR" --strip-components=1; \
	rm "$tarball" "$tarball.asc";
```

ecdye, 2022-03-09:
No it is not try using <https://keys.openpgp.org/> as your key server.

closed 2022-03-09 by ecdye