ecdye/zram-config #97: Allow update.bash without git fetch/pull

strayer, 2022-07-29:
I'm using Ansible to install zram-config and set a specific git tag I'd like to check out (e.g. v1.5.0). I configured Ansible to run update.bash when it changes the checked out git repo (e.g. when I change the Ansible config from v1.4.0 to v1.5.0).

Problem is that update.bash always fetches and checks out the main branch. I'd like to be able to just let it update the installed files after checking out a branch/tag myself.

Thanks for your work!

tux-ware, 2022-09-30:
+1 for having the possibility to check out specific branch and use it within the update.bash script.

tux-ware, 2022-12-01:
This is the only issue reported at the moment. Any update on the current status?

ecdye, 2024-05-27:
Sorry for the lack of response on this, I have been away serving as a missionary. This is a great suggestion, and I will begin working on this soon.

closed 2024-05-27 by ecdye

strayer, 2024-05-27:
No need to apologize, thank you for looking into this and still maintaining the project 🎉

commit 5867021c (2024-05-27): Add custom changes functionality to update.bash

Fixes #97