ecdye/zram-config #100: F2FS filesystem compatibility

sulisu, 2023-03-05:
I have F2FS on my flash driver and zram-config failed to work. It seems that the code is hard coded to work with ext3 or ext4 filesystem.

I replaced this line

```
mke2fs -v -t "$dirFSType" "/dev/zram${RAM_DEV}" &>> "$ZLOG" || return 1
```

to

```
	if [[ $dirFSType == "ext4" ]]; then
		mke2fs -v -t "$dirFSType" "/dev/zram${RAM_DEV}" &>> "$ZLOG" || return 1
	elif [[ $dirFSType == "f2fs" ]]; then
		mkfs.f2fs "/dev/zram${RAM_DEV}" &>> "$ZLOG" || return 1
	fi
```

Now it works.

tux-ware, 2023-05-10:
Thanks for sharing how to add F2FS support.

ecdye, 2024-06-22:
I like the idea behind this, but I would like to explore it more, and see if we could add support for any filesystem type by identifying it dynamically.

commit c9debbbb (2024-06-22): Add support for alternate filesystems

Fixes #100
PR #111 (2024-06-22): Add support for alternate filesystems
Fixes #100
commit e1370eeb (2024-06-22): Add support for alternate filesystems (#111)

Fixes #100
closed 2024-06-22 by ecdye