ecdye/zram-config #116: No logs after launching zram

skibbipl, 2024-07-28:
So I decided to experiment on a VM with zram config. I installed debian testing and installed zram-config. Installation went smoothly without errors but journalctl stopped working. What I can see after reboot is:
```
jul 28 22:17:45 temp systemd-journald[286]: Journal stopped
```
Am I missing something? I expected that logs will work without issues, but it seems that there is no logging at all after installing zram-config. My ztab is unchanged:
```
log     lzo-rle         50M             150M            /var/log                /opt/zram/log.bind      /opt/zram/oldlog
```
Trying to enable sytemd-journald ends with error:
```
root@temp:~# service systemd-journald start
A dependency job for systemd-journald.service failed. See 'journalctl -xe' for details.
```

ecdye, 2024-07-30:
One issue you are probably going to run into is that generally speaking zram won't work properly in a VM. This is one of the reasons that I use a custom emulated image for my testing suite on GitHub.

Have you tried an install on a non VM and observed the same issue?

skibbipl, 2024-07-30:
Thanks for pointing this out - perhaps it's worth updating known issues part of the documentation?
I will check it on my raspberry pi, however I wanted to implement it on proxmox node which runs temporarily on pendrive.

ecdye, 2024-08-06:
For sure, I'll look into adding a note about that. Thanks for the feedback!

PR #117 (2024-08-06): Add note about VM compatibility to README.md
Fixes #116
commit 203ae245 (2024-08-06): Add note about VM compatibility to README.md (#117)

Fixes #116
closed 2024-08-06 by ecdye