ecdye/zram-config #121: compile warnings installing zram-config

ArjenR49, 2024-09-15:
Warnings like these are enough to send me up the wall ... installing some Meson build system and ninja? 
I was only looking to install zram-config, not a compiling system and what have you.
Are these errors bad?
Can I use zram-config on  my Pi Zero 2 PiHole/Unbound/DNS and local NTP server as it is now with those errors?
It runs Raspberry Pi OS Bookworm 32 bits.

Perhaps there is some syntax error that I can put right by editing install.bash?
Github is strange to me. Obviously very useful, if you have studied it first.

Not too long ago I installed zram-config on another Pi Zero and there was no --recurse-modules on the download command AND no errors, either. In that log there's not a trace of Meson system and Ninja. (Ninja sounds silly science fiction, btw. Probably from some TV program for children .... except my other R7800 router has a Ninja logo ... )

`pi@PiHole3:~ $ sudo ./zram-config/install.bash
The Meson build system
Version: 1.0.1
Source dir: /home/pi/zram-config/overlayfs-tools
Build dir: /home/pi/zram-config/overlayfs-tools/builddir
Build type: native build
Project name: overlayfs-tools
Project version: 2024.07
C compiler for the host machine: cc (gcc 12.2.0 "cc (Raspbian 12.2.0-14+rpi1) 12.2.0")
C linker for the host machine: cc ld.bfd 2.40
Host machine cpu family: arm
Host machine cpu: armv7l
Found pkg-config: /usr/bin/pkg-config (1.8.1)
Run-time dependency musl-fts found: NO (tried pkgconfig)
Library m found: YES
Program test_cases/run_tests.py found: YES (/home/pi/zram-config/overlayfs-tools/test_cases/run_tests.py)
Build targets in project: 10

Found ninja-1.11.1 at /usr/bin/ninja
INFO: autodetecting backend as ninja
INFO: calculating backend command to run: /usr/bin/ninja -C /home/pi/zram-config/overlayfs-tools/builddir
ninja: Entering directory `/home/pi/zram-config/overlayfs-tools/builddir'
[5/13] Compiling C object fsck.overlay.p/fsck.c.o
In file included from ../fsck.c:42:
../fsck.c: In function ‘ovl_open_dirs’:
../common.h:32:17: warning: format ‘%lu’ expects argument of type ‘long unsigned int’, but argument 2 has type ‘rlim_t’ {aka ‘long long unsigned int’} [-Wformat=]
   32 | #define _(x)    (x)
      |                 ^~~
../fsck.c:71:28: note: in expansion of macro ‘_’
   71 |                 print_info(_("Process fd number limit=%lu "
      |                            ^
../common.h:32:17: warning: format ‘%lu’ expects argument of type ‘long unsigned int’, but argument 3 has type ‘rlim_t’ {aka ‘long long unsigned int’} [-Wformat=]
   32 | #define _(x)    (x)
      |                 ^~~
../fsck.c:71:28: note: in expansion of macro ‘_’
   71 |                 print_info(_("Process fd number limit=%lu "
      |                            ^
[13/13] Linking target fsck.overlay
ninja: Entering directory `/home/pi/zram-config/overlayfs-tools/builddir'
ninja: no work to do.
Installing overlay to /usr/local/bin
Installing fsck.overlay to /usr/local/bin
Installing zram-config files
Starting zram-config service
Created symlink /etc/systemd/system/multi-user.target.wants/zram-config.service → /etc/systemd/system/zram-config.service.
#####     zram-config is now installed and running     #####
#####       edit /etc/ztab to configure options        #####
pi@PiHole3:~ $ 

`

Thanks for the good work, not freaking out unsuspicious users ....
Thanks again!!!

ecdye, 2024-09-16:
Hi @ArjenR49,

This is normal and expected behavior. I recently switched to using a version of overlayfs-tools that is managed in another repository. It is a dependency for zram-config, and the repository I switched to uses Meson as a build system with `ninja` being the compiler frontend that is used.

I'm sorry if this alarmed you, but it's nothing to worry about. The warnings you are seeing are simply normal compile-time warnings displayed by the compiler for the benefit of the developer.

Let me know if you have any questions.

closed 2024-09-16 by ecdye

ArjenR49, 2024-09-16:
Thanks for your swift reply.
I've had similar compile warnings also when installing a simple utility for a PiAware ADS-B SDR receiver module. It did work and switched on supply current to the antenna side of the module, so I decided to forget about it. That isn't anything critical. The DNS server, however, is pretty critical to my LAN. I'm glad you seem to appreciate my worries.
I suppose it could help other, too, to read there's nothing to worry about.